mantidproject/mantid #23164: Plot result crashes MSD in Indirect Data Analysis

keeeto, 2018-08-02:
**Original reporter:** Sanghamitra

### Expected behavior

Clicking Plot Result after a fit should produce a plot

### Actual behavior

Mantid throws an uncaught exception and does not plot

### Steps to reproduce the behavior

* Open Mantid
* Open Indirect Data Analysis window
* Navigate to the MSD tab
* Load the data below
* Choose `Fit type` Gaussian
* Click`Run`
* After the fitting is finished click `Plot Result`

### Platforms affected

Windows, OSX, probably all

SpencerHowells, 2018-08-02:
@AnthonyLim23 @robertapplin If the data only has 1 spectrum then the plot option should be disabled.
It is to be used for several spectra which correspond say to temperature variation as in
[osi104367-104371_graphite002_red_elwin_eq.zip](https://github.com/mantidproject/mantid/files/2253743/osi104367-104371_graphite002_red_elwin_eq.zip)
The Plot then gives parameters as function of temperature. However the *_result WS has incorrect x-unit it should be that in the *_eq WS in this case temperature not Q.

AnthonyLim23, 2018-08-02:
@SpencerHowells I discussed this with Keith earlier, I think it would be better to just hide the option altogether. Otherwise the user may get confused as to why the plot single spectrum is unticked and they cannot tick it.

AnthonyLim23, 2018-08-02:
This is part of umbrella issue #23156

SpencerHowells, 2018-08-02:
@AnthonyLim23 We have several issues here. The input *_eq may have a single spectrum or many - in the latter case the plot and fit are definitely needed. The file attached above will show this & the plot works. This file
[osi92762_graphite002_elwin_eq.zip](https://github.com/mantidproject/mantid/files/2254398/osi92762_graphite002_elwin_eq.zip)
is a single spectrum and Fit Single Spectrum works (for me anyway).

closed 2018-09-28 by gvardany