mantidproject/mantid #23183: Tiled plot crashes in I(QT) Widget of Indirect Data Analysis

keeeto, 2018-08-03:
**Original reporter:** Saghamitra

### Expected behavior

Mantid should not freeze or crash.

### Actual behavior

Mantid freezes or crashes

### Steps to reproduce the behavior

* In Indirect > Data Analysis go to MSD tab
* Load data `iris26184_multi_graphite002_elwin_eq.nxs` - available in the zip below
* Choose Fit Type = Gaussian
* Choose Run
* After the run choose Plot Data
* This should result in an uncaught exception error - this is documented in another issue (#23164)
* Choose `Do not share information` and `Continue working`
* Go to the I(Q,T) window and load
    * Sample: `iris26184_multi_graphite002_red.nxs`
    * Resolution: `irs26173_graphite002_res.nxs`
* Choose Run - this process gets stuck and Mantid hangs
* Note the process does not hang if one does not have the first error in the MSD interface

[Archive.zip](https://github.com/mantidproject/mantid/files/2257638/Archive.zip)

### Platforms affected

All

SpencerHowells, 2018-08-03:
@keeeto @robertapplin what happens when you run I(q,t) on its own? Ie not after a crash. This should establish whether it is a problem in I(q,t) or a consequence of the crash.

keeeto, 2018-08-03:
@SpencerHowells - it runs fine without the previous crash. 
I think that fixing #23164 should cover this one too.

robertapplin, 2018-11-27:
This has been fixed by #23441

closed 2018-11-27 by robertapplin