mantidproject/mantid #23229: Plot Guess causes uncaught exception

keeeto, 2018-08-09:
**Original reporter:** n/a

### Expected behavior

Plot guess should show a guess of the function to be fitted.

### Actual behavior

For certain functions no guess is shown and Mantid throws an uncaught exception

### Steps to reproduce the behavior

* Go to `Interfaces` > `Indirect` > `Data Analysis`
* Go to the `Conv Fit` tab
* Load the `irs26174_graphite002_red.nxs ` file from the sample data
* Load the resolution file `irs26173_graphite002_res.nxs` from the sample data
* Check the `Plot Guess` box
* Set the function to InelasticDiffRotDiscreteCircle
* Mantid throws an uncaught exception

### Platforms affected

OSX at least

SpencerHowells, 2018-08-09:
@keeeto @AnthonyLim23 The *diff* functions together with TeixieraWater are SNS written code to fit in S(q,w). To do this they need to read the q value from the workspace histograms. This functionality needs to be sorted out. I have tried testing them & no user has tried them either. Now that Brandon has sorted out TeixieraWater the others need to,be looked at in detail.

keeeto, 2018-08-10:
Hi @SpencerHowells thanks for the input. Do you mean that the `InelasticDiffRotDiscreteCircle` does not function properly with reading the q-values from workspace histograms? Also are you saying that nobody is using this functionality at the moment?

SpencerHowells, 2018-08-11:
@keeeto I’ll have another look at these. If I remember correctly they work after Dan did a fix. However there is a better method which Jose wrote for TeixieraWater. These old fixes need to be updated to the new method. I don’t know of any ISIS user trying it.

PR #23453 (2018-09-12): Prevent Plot Guess Crash on ConvFit by Reading Q Values
**Description of work.**
This PR prevents a crash when Plot Guess is clicked for the Fit Types `InelasticDiffSphere` and `InelasticDiffRotDiscreteCircle`. The Q values were missing, and so they had to be retrieved from the workspaces in order to fix this (like with `Teixeira Water`).
This PR also makes sure the Q values for `ElasticDiffSphere` and `ElasticDiffRotDiscreteCircle` are retrieved from the workspace.

**To test:**
1. `Indirect`->`Data Analysis` ->`ConvFit` Tab
2. Load in the data below
3. Select `InelasticDiffSphere` as Fit Typ [... cut]
closed 2018-09-12 by soininen