mantidproject/mantid #23238: Remove non-sync scan from DetectorInfo

OwenArnold, 2018-08-10:
[DetectorInfo](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/inc/MantidGeometry/Instrument/DetectorInfo.h#L125) currently supports two modes of scanning detectors.

1. Time intervals are bespoke per detector. 
2. Time intervals are global. This is the synchronised (sync) scan.

**We only need to support the latter**. The ILL do not use the former, and the latter is more memory efficient. This means that bits of the API such as [this](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/inc/MantidGeometry/Instrument/DetectorInfo.h#L125) and the mechanisms behind them can be removed. Remember to remove from both `Geometry::DetectorInfo` and `Beamline::DetectorInfo` [here](https://github.com/mantidproject/mantid/blob/master/Framework/Beamline/inc/MantidBeamline/DetectorInfo.h)

@SimonHeybrock might be able to provide more detail on this. We should also get a review from the ILL on this.

SimonHeybrock, 2018-08-13:
@OwenArnold Do we have a solution for the case of monitor data that does not distinguish between scan intervals? Do we need a solution?

OwenArnold, 2018-08-13:
@SimonHeybrock good point, we probably need one. I'm not sure that one exists.

PR #23744 (2018-10-22): Remove non-sync scan feature from DetectorInfo.
**Description of work.**

DetectorInfo has 2 modes for scanning detectors:
1. Individual time intervals per detector (non-sync scan)
2. Global intervals for all detectors (sync scan)
We only need the second version, which is also simpler and can lead to optimisations.

I removed all places in DetectorInfo and ComponentInfo where non-sync scan was used.
`m_scanCounts` and `m_scanIntervals` are now initialised to
```
size_t m_scanCounts = 1;
std::vector<std::pair<int64_t, int64_t>> m_scanIntervals{{0, 1}};
```
instead of being `null` pointers.

In  [... cut]
closed 2018-10-22 by AndreiSavici