mantidproject/mantid #23308: Monitor should be spcified by number i.e. M3/M4  not spectrum number

Mantid-Matthew, 2018-08-21:
The monitor to use for a reduction should be specified by selecting M3/m4 rather than a spectrum number as several of the monitor spectrum numbers are odd and hard to remember.

Mantid-Matthew, 2018-09-25:
Having investigated this I need to speak to the scientists some more. The transmission spectrum is specified in the user file. To convert this to a monitor number I then need to know the monitor to spectrum mapping. You can't know for certain the monitor to spectrum mapping until after some data is loaded however as you need a date to know which IDF to use.

smk78, 2020-07-16:
On review, several members of the SANS Group expressed concern about this and would like more more information about how it is planned to be implemented. What happens if they change monitor spectrum numbers, or if they use a ROI of several spectra for the transmissions, for example?

smk78, 2021-02-23:
Requires further discussion between the SANS Group & @DavidFair

closed 2021-08-29 by stale[bot]