mantidproject/mantid #23446: Wrong assumption about IDF name

AndreiSavici, 2018-09-05:
I get a warning when I am building documentation due to 
```
IndirectDataReduction-[Warning] Failed to load instrument with error: FileDescriptor() - File '/home/3y9/Mantid/mantid/instrument/BASIS_Definition.xml' does not exist. The current facility may not be fully supported.
IndirectDataReduction-[Warning] Instument loading failed! This instrument (or analyser/reflection configuration) may not be supported by the interface.
```
In two of the files, qt/scientific_interfaces/Indirect/IndirectDiffractionReduction.cpp and qt/scientific_interfaces/Indirect/IndirectDataReduction.cpp, one makes assumption that the IDF name is just `instrumentName+"_Definition.xml"`. For BASIS you also have a date range.

**Original reporter:** [username facility]/[nobody]. 

### Steps to reproduce the behavior
Set facility to SNS, then build documentation.
### Platforms affected
All

robertapplin, 2019-02-12:
@AndreiSavici Do you happen to know how it is decided which date range should be used? There are seem to be three definition files: `BASIS_Definition_0-20130119`, `BASIS_Definition_20130121-20131231` and `BASIS_Definition_20140101-`

AndreiSavici, 2019-02-12:
The last one. The IDF name contains `valid from - valid to` dates.

closed 2019-07-01 by robertapplin