mantidproject/mantid #23471: ReflectometryReductionOneAuto2 Throws error for singular transmission runs being passed, expecting groups.

Pasarus, 2018-09-10:
### Expected behavior
To not throw: 
"Property "FIRSTTRANSMISSIONRUN" is not set to a valid value: "Workspace OFFSPEC44937_1 is not of the correct type"."

### Actual behavior
It throws:
"Property "FIRSTTRANSMISSIONRUN" is not set to a valid value: "Workspace OFFSPEC44937_1 is not of the correct type"."

### Steps to reproduce the behavior
Run this:
OFFSPEC44956 = LoadISISNexus(Filename="OFFSPEC44956")
OFFSPEC44937 = LoadISISNexus(Filename="OFFSPEC44937")
ReflectometryReductionOneAuto(InputWorkspace='OFFSPEC44956', ThetaIn=0.7, 
  MomentumTransferStep=0.03, MomentumTransferMax=0.1, 
  OutputWorkspaceBinned='IvsQ_binned', OutputWorkspace='IvsQ', OutputWorkspaceWavelength='IvsLam',
  FirstTransmissionRun="OFFSPEC44937", AnalysisMode="MultiDetectorAnalysis",
  ProcessingInstructions="401-411")

### Platforms affected
All

Tips:
The actual error is decided at line: 443 in PropertyManager.cpp in the isValid() method.
The isValid() method calls the WorkspaceProperty.tcc file at line 201, furthermore look at line 246 for the actual decision that causes the error, it seems to be expecting a group when we only need a MatrixWorkspace.

gemmaguest, 2019-11-14:
I wonder if this could be related to the fact that the base class `initTransmissionProperties` declates the input properties as `MatrixWorkspace` for both RRO and RROA whereas RROA may have `WorkspaceGroup`s as inputs. The error would seem to be the wrong way round for this, although I can't see exactly when the error is being thrown, so it's worth looking at.

gemmaguest, 2020-09-10:
This keeps coming up in unscripted testing. It may be a spurious error but we should check it's not a symptom of something more serious. It would be good to fix it either way, but if it turns out to be a time consuming fix for something we can safely ignore we should at least update the test instructions to avoid confusion in future: http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

DannyHindson, 2020-09-10:
Had a look into this. The problem comes from ReflectometryReductionOneAuto3::processGroups(). If the input workspace and the transmission workspaces provided are all group workspaces then the first workspace in the group is always used as processGroups steps through the different workspaces in the main input workspace group. Unfortunately the transmission workspace gets cleared at the end of the first iteration by this code in Algorithm.cpp:
```
      // Put the output workspaces into the AnalysisDataService - if requested
      if (m_alwaysStoreInADS)
        this->store();
```

store() stores any output workspaces in the ADS but it also clears all workspaces (both input and output). Unclear whether it should only be clearing the output workspaces. Will check separately. Doesn't affect the results because Algorithm has a 2nd stab at loading any workspace properties if they're initially empty so downside of this is just red error messages in log

PR #29822 (2020-11-18): Fix spurious transmission error with workspace groups
**Report to:** Max at ISIS

**Description of work.**
- This PR fixes some spurious error messages that are issued when you run ReflectometryReductionOneAuto on a workspace group, with transmission corrections enabled. The problem was related to running the same child algorithm multiple times. Now, a new child algorithm is created each time.
- To do this some refactoring was required. Other tidying has also been done to simplify the `processGroups` function and reduce duplicated code
- The documentation regarding processing groups has also b [... cut]
closed 2020-11-18 by martyngigg