mantidproject/mantid #23488: Change license in source files

peterfpeterson, 2018-09-11:
Changing how license is specified in headers to follow ([spdx](https://spdx.org/licenses/) following [examples](https://spdx.org/sites/cpstandard/files/pages/files/using_spdx_license_list_short_identifiers.pdf)).

peterfpeterson, 2018-09-13:
The proposed text for the license/copyright should be stated in the "details" for this issue before implementing.

NickDraper, 2018-09-17:
Example of a current header

/** ArrayLenghtValidator : Validate length of an array property

  Copyright &copy; 2011 ISIS Rutherford Appleton Laboratory, NScD Oak Ridge
  National Laboratory & European Spallation Source

  This file is part of Mantid.

  Mantid is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 3 of the License, or
  (at your option) any later version.

  Mantid is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.

  File change history is stored at: <https://github.com/mantidproject/mantid>
  Code Documentation is available at: <http://doxygen.mantidproject.org>
*/

NickDraper, 2018-09-17:
Proposed adjusted header for .h

/** ArrayLengthValidator : Validate length of an array property

Mantid Source Repository: https://github.com/mantidproject/mantid

Copyright © 2011 ISIS Rutherford Appleton Laboratory UKRI, NScD Oak Ridge
  National Laboratory, European Spallation Source & Institut Laue-Langevin

SPDX-License-Identifier: GPL-3.0+
*/

NickDraper, 2018-09-17:
Do we need a year range in the copyright statement?

No. Assuming the code is created and published (posted to Github) within 25 years, then it is only the date of first publication that matters.

Quoted from https://www.copyright.gov/circs/circ15a.pdf (US Law)
> For works made for hire and anonymous and pseudonymous works, the duration
> of copyright is 95 years from first publication or 120 years from creation,
> whichever is shorter

NickDraper, 2018-09-17:
Option 2: Split out a file header with just the copyright and license info

/* 
 * Copyright &copy; 2011 ISIS Rutherford Appleton Laboratory UKRI, NScD Oak Ridge
 *     National Laboratory, European Spallation Source & Institut Laue - Langevin
 *
 * SPDX - License - Identifier: GPL - 3.0 +
 *
 * Mantid Repository : https://github.com/mantidproject/mantid
 */

Then have class headers as pure doxygen
/** Validates the length of an array property
*/

PR #23693 (2018-10-05): Update file headers, new copyright details
**Description of work.**
This is an update of the copyright statements in Mantid.

As well as updating the comment in .h files, it adds it to all of the cpp and py files as well, where many are missing.

This is done via a new script in Tools/Copyright/copyrightupdate.py,  The script detects and updates both new and old style copyright statements, so can handle changes to the statement in the future.

**Report to:** nobody

**To test:**

Fixes #23488. 

This does not require release notes because it is purely internal to our code.

---

#### Reviewer  [... cut]
closed 2018-10-05 by martyngigg