mantidproject/mantid #23579: Failing system tests

peterfpeterson, 2018-09-20:
#23573 highlighted that there are [7 broken system tests in `master`](http://builds.mantidproject.org/job/pull_requests-rhel7/23567/testReport/).

Most the failures are related to the changes introduced in #23514 as verified by reverting
```
git revert  2b89a12291cc9f7fc64fa103835ed68ae7cc8724 -m 1
```

* SANS2DMultiPeriod.LARMORMultiPeriodEventModeLoading
* ISIS_PowderPearlTest.CreateVanadiumAllTest
* ISIS_PowderPearlTest.CreateVanadiumGroupsTest
* ISIS_PowderPearlTest.CreateVanadiumModsTest
* ISIS_PowderPearlTest.CreateVanadiumTransTest
* ISIS_PowderPearlTest.FocusWithAbsorbCorrectionsTest
* SingleCrystalDiffuseReductionTest.SingleCrystalDiffuseTest

The one that I haven't figured out the cause of is (the previous revert does not fix the test)
* ISIS_PowderGemTest.FocusTest

peterfpeterson, 2018-09-20:
The problem with the Pearl algorithms is in [`pearl_algs.apply_vanadium_absorb_corrections`](https://github.com/mantidproject/mantid/blob/master/scripts/Diffraction/isis_powder/pearl_routines/pearl_algs.py#L33) the absorption workspace has different binning for each spectrum. The old implementation of `RebinToWorkspace` only looked at the first spectrum of `WorkspaceToMatch` and converted everything to that binning. The current implementation goes through each spectrum of `WorkspaceToRebin` and `WorkspaceToMatch` in parallel and matches them. `Divide` doesn't notice that the x-axes are different values since [`BinaryOperation::checkCompatibility`](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/BinaryOperation.cpp#L304) only checks the units and length of the x-axis rather than the x-values.

This appears to have exposed a bug (or not well documented behavior). Someone who knows the instrument better (and can talk with the affected staff) needs to decide what the right thing to do is. The two options are:

**Keep the new behavior and update the reference files**

 In my opinion the current implementation is still wrong. It *should* be that one rebins the absorption workspace to match the data, divide and delete. In other words, the implementation should be
```python
def apply_vanadium_absorb_corrections(van_ws, run_details, absorb_ws=None):
    if absorb_ws is None:
        absorb_ws = mantid.Load(Filename=run_details.vanadium_absorption_path)

    van_original_units = van_ws.getAxis(0).getUnit().unitID()
    absorb_units = absorb_ws.getAxis(0).getUnit().unitID()
    if van_original_units != absorb_units:
        van_ws = mantid.ConvertUnits(InputWorkspace=van_ws, Target=absorb_units, OutputWorkspace=van_ws)

    absorb_ws = mantid.RebinToWorkspace(WorkspaceToRebin=absorb_ws, WorkspaceToMatch=van_ws, OutputWorkspace=absorb_ws)
    van_ws = mantid.Divide(LHSWorkspace=van_ws, RHSWorkspace=absorb_ws, OutputWorkspace=van_ws)

    if van_original_units != absorb_units:
        van_ws = mantid.ConvertUnits(InputWorkspace=van_ws, Target=van_original_units, OutputWorkspace=van_ws)

    common.remove_intermediate_workspace(absorb_ws)
    return van_ws
```
Since the all of the reference files are being updated already with this choice, it can be done all the same time.

**Be explicit about matching the binning of the first spectrum in the absorption workspace**

This would be done by copying the code from the  [update to `IndirectReductionCommon.py`](https://github.com/mantidproject/mantid/pull/23514/files#diff-aedc947fbca20a5709ceda136dcb97fdL991). If this option is selected, then there needs to be a comment on that decision in the code so future developers don't break it.

peterfpeterson, 2018-09-20:
I'm suspicious that the issue with ISIS_PowderGemTest.FocusTest derives from `RebinToWorkspace` as called in [`focus._divide_one_spectrum_by_spline`](https://github.com/mantidproject/mantid/blob/master/scripts/Diffraction/isis_powder/routines/focus.py#L118) but that doesn't explain why reverting the changes from #23514 didn't fix it.

peterfpeterson, 2018-09-21:
[The list](http://builds.mantidproject.org/view/Master%20Pipeline/job/master_systemtests-rhel7/744/) is shorter now.

SamJenkins1, 2018-09-22:
My PR passes the gem test, though still fials system tests due to the other 5 tests