mantidproject/mantid #23628: Expand code completion in workbench

peterfpeterson, 2018-09-26:
**Original reporter:** [username facility]/[nobody]. 

### Expected behavior

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

NickDraper, 2019-09-02:
From another issue closed as a duplicate

The following script will give an intellisense list of operations in mantidplot, but not the workbench.

We should make them equal
```
ws = CreateSampleWorkspace()
ws.
```

NickDraper, 2019-09-24:
Additional test case:

Expected behavior
def func((a,b)):
    pass
When run from code editor should exit cleanly

Actual behavior
Script appears to run but then crashes code editor tab with error:

Traceback (most recent call last):
  File "C:\MantidInstall\bin\mantidqt\widgets\codeeditor\interpreter.py", line 332, in _on_exec_success
    self.view.editor.updateCompletionAPI(self.model.generate_calltips())
  File "C:\MantidInstall\bin\mantidqt\widgets\codeeditor\execution.py", line 184, in generate_calltips
    calltips.append(name + get_function_spec(attr))
  File "C:\MantidInstall\bin\mantidqt\widgets\codeeditor\execution.py", line 80, in get_function_spec
    calltip = ','.join(args)
TypeError: sequence item 0: expected string, list found
Steps to reproduce the behavior
Enter

def func((a,b)):
    pass
into code editor and run

Platforms affected
I’m running in workbench 4.1.0 on Windows 10

closed 2019-10-18 by gemmaguest