mantidproject/mantid #23634: Workbench does not run scripts a second time

peterfpeterson, 2018-09-26:
In the editor window create a script
```python
print('hello world!')
```
Then press ctrl-enter. In the log window you'll see "hello world!". Press ctrl-enter again and you won't see an update in the log window.

peterfpeterson, 2018-09-26:
Now it appears to be working fine.

peterfpeterson, 2018-09-26:
Editing the script and executing it again doesn't fire. However, adding a print statement in `MultiPythonFileInterpreter.execute_current()` fixes it.

martyngigg, 2019-01-14:
Antti found this to be problematic:
```python
from __future__ import (absolute_import, division, print_function, unicode_literals)

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *

import matplotlib.pyplot as plt

import numpy
import directtools

ws = CreateSampleWorkspace(NumBanks=1, XUnit='DeltaE', XMin=-3., XMax=12., BinWidth=0.1, StoreInADS=False)
ws = ConvertSpectrumAxis(ws, 'Theta', 'Direct', 4., StoreInADS=False)
stw = ComputeIncoherentDOS(ws, StoreInADS=True)
kwargs = {'workspaces': stw, 'xscale': 'log', 'yscale': 'log'}
figure, axes =directtools.plotDOS(**kwargs)
```
Execute, change one True to False, execute again, change back, ... It starts to fail to execute pretty quickly for me.
...it seems to be connected with the autofill widget. If it opens during writing some code, and I try to execute while the widget is open, the execution will become blocked.

peterfpeterson, 2019-01-14:
@DTasev "... found that if I open a new editor, then delete all the text, press ctrl+enter to run it, nothing happens (as expected), but then after typing in new code it cannot be executed either."