mantidproject/mantid #23642: Muon - Muon Analysis 2.0 Home Tab, Grouping Tab, Load Widget

TWJubb, 2018-09-27:
This issue will be attached to a branch containing the Home Tab, Grouping Tab and Load Widgets for the new Muon analysis interface. 

Smaller branches will be created from this branch to hold small PR's which will be merged sequentially into master.

List of PRs
===========

-

keeeto, 2018-10-02:
After spending some time testing this I think that on the whole it works nicely. There are a few issues that could be cleared up

- [ ] Blue lines on interface in OSX (see figure at the end)
- [ ] The deadtime printed on the screen does not update after changing the loaded file
- [ ] An 'are you sure' dialog would be useful for the clear buttons
- [ ] Grouping tab allows numbers outside the range, this causes an error in the log, but remains in the GUI
- [ ] Failure to save a grouping file does not throw an error dialog
- [ ] Saving the grouping file requires explicit inclusion of `.xml`, this should be default
- [ ] Changing the value of alpha in the grouping tab switches `group/pair` from 'long' back to 'fwd' in the main window
- [ ] Load current run could be greyed out when unavailable
[image]

TWJubb, 2018-10-10:
Comments from an instrument scientist : 

- [ ] GUI automatically sets splitter to minimize the empty vertical space.
- [ ] GUI opens with horizontal space automatically minimized.
- [ ] Bug : Instrument combo box should change to newly selected instrument immediately (rather than waiting for data to be loaded).

closed 2019-02-08 by AnthonyLim23