mantidproject/mantid #23669: Update system tests for WISH

samueljackson92, 2018-10-01:
We should write/update the system tests for WISH data:
 - [ ] write a test for the autoreduction powder diffraction script
 - [ ] write a test which is more representative of the single crystal reduction script.
 - [ ] write additional tests for covering variations of how they run
    - [ ] double frame mode
    - [ ] slicing mode

SamJenkins1, 2018-10-04:
Where are the wish scripts, I was going to look into this, and then realised I have no idea where the wish scripts are.

ElliotAOram, 2018-10-04:
@SamJenkins1 sorry didn't see your comment. This one's all yours! The autoreduction script is `\\isis\inst$\NDXWISH\User\scripts\autoreduction\reduce.py` hopefully @samueljackson92 can point you to the other scripts

SamJenkins1, 2018-10-04:
no problem, I should have self-assigned it when sam messaged me with the details

SamJenkins1, 2018-10-10:
Just though I should leave a note here while I thought of it, if anyone is gonna try tackling the SX tests, running the SX script on my computer ran out of memory, so you'll want a computer with more than 32 gigabytes of RAM

SamJenkins1, 2018-10-15:
I'm nursing through the powder reduction test now, but I don't have enough ram on my computer to test SX, would someone else mind taking over that one?

closed 2021-03-02 by stale[bot]