mantidproject/mantid #23725: Preferences GUI for new workbench

AndreiSavici, 2018-10-04:
With a clean mantid (no settings file), the default facility is not set. Some algorithms (e.g. DgsReduction) require the facility to be set.

To test: delete the `.mantid` folder in the user directory (or the equivalent on operating systems other than linux), and run `DgsReduction` on the `CNCS_7860` data from doctests.

martyngigg, 2019-01-28:
Lets start with some minimal options organized into some sensible pages on the dialog
* facility/instrument selection
* visible algorithms/interfaces
* ask on close for saving projects

Spyder has quite a nice page-per-area [image]

martyngigg, 2019-01-28:
This could also be a good place to put things like the default editor content selection in #24580.

PR #24790 (2019-02-27): Settings window for the Workbench
**Description of work.**
Adds a Settings module for the workbench.

**To test:**
Open File > Settings, or press Ctrl+Alt+S.

Fixes #23725 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolation?
- If there are changes in the release notes then do they descr [... cut]
closed 2019-02-27 by peterfpeterson