mantidproject/mantid #23767: Muon Thread model

AnthonyLim23, 2018-10-11:
In PR #23763 the thread model was reverted to prevent crashes of muon interfaces (frequency domain analysis and elemental analysis) on master. Below are the steps to cause one of the crashes (on linux): 

Open muon analysis and load some data
Then open the frequency domain analysis GUI
Change to maxent
Tick the use phases box in the table
Click the calculate button - it should give a warning box and not crash

TWJubb, 2018-10-11:
I will check this out straight away; I have a Linux build building. I may ask for the error report from the log if I can't reproduce it... I assume there were some python-esque errors showing?

TWJubb, 2018-10-11:
This can be fixed in a single line;

The `ThreadModel` class should inherit from `QtGui.QWidget` rather than `QtCore.QObject`.

TWJubb, 2018-10-11:
Just to be certain, the error box contains the text

```
Cannot currently generate phase table from this data using CalMuonDetectorPhases
```

is that correct?

AnthonyLim23, 2018-10-11:
yep, that is the correct error message

closed 2019-01-08 by AnthonyLim23