mantidproject/mantid #23808: DirectILL: S(2theta, TOF)

soininen, 2018-10-15:
**Original reporter:** Jacques Ollivier/ILL

For visualization purposes, the IN5 scientists would like to conveniently get the S(2theta, TOF). This could be realized by an algorithm which does the following steps:
1. `ConvertToConstantL2`
2. `ConvertSpectrumAxis` (spectrum numbers to theta)
3. `GroupDetectors` (group over Debye-Scherrer cones, width should be user-specifiable, defaulted to a nice instrument specific value)

AndreiSavici, 2018-10-15:
Note that there is the [GenerateGroupingPowder](http://docs.mantidproject.org/nightly/algorithms/GenerateGroupingPowder-v1.html) that you might be able to use

soininen, 2018-10-15:
@AndreiSavici I just remembered a similar commit from you on another issues and was going to mention `GenerateGroupingPowder` to the description :) I'll have a look at it, thanks for the reminder!

PR #24030 (2018-11-27): New algorithm: SofTwoThetaTOF
This PR introduces a new algorithm `SofTwoThetaTOF` which averages a direct geometry time-of-flight workspace over constant 2theta producing an S(2theta, TOF) workspace. The instrument scientists are intending to use this type of workspace for diagnostic purposes at the IN5 instrument at the ILL.

This PR also makes the generation of the `.par` file in `GenerateGroupingPowder` optional as the file is not needed for `SofTwoThetaTOF` and would complicate the cleanup process.

**Report to:** Jacques Ollivier/ILL.

**To test:**

Ensure that the unit test data director [... cut]
closed 2018-11-27 by martyngigg