mantidproject/mantid #23817: Crop before stitching results of RROA

gemmaguest, 2018-10-16:
**Original reporter:** Jos at ISIS

When processing a group of runs with the ISIS Reflectometry GUI, the individual run outputs from `ReflectometryReductionOneAuto` should be cropped before being stitched. Note however that they shouldn't be rebinned, so we can't just use the OutputWorkspaceBinned output from RROA.

It would be easiest if we could change the `OutputWorkspace` from RROA and RRO to always be cropped - check with Jos whether this is what they want.

gemmaguest, 2019-03-14:
This has just been highlighted by scientists as being higher priority so I would like to take a look to see if we can get it into this release, if it's low risk (which I expect it to be)

NickDraper, 2019-03-21:
Gemma can you make sure Jos knows that this is included in the release

PR #25199 (2019-03-21): Crop the IvsQ output of RROA if Q limits specified
If the momentum transfer min and/or max are specified, crop the IvsQ output workspace to these limits. This is an interim output workspace which is used further down the reduction for stitching, and it is important to crop unwanted regions before the stitch. The main output (IvsQ_binned) is already cropped as a result of the rebin step.

Fixes #23817

**Report to:** Max, Jos and Christy at ISIS

**To test:**
This script runs various combinations of setting the min/max and also checks that it works with/without specifying the rebin step.
```
Lo [... cut]
closed 2019-03-21 by NickDraper