mantidproject/mantid #23903: ISIS reduction always crashes

abuts, 2018-10-29:
**Original reporter:** [ISIS excitation group]/[DaaaS testers]. 
Att of: @martyngigg; @NickDraper 

### Expected behavior

### Actual behavior
ISIS reduction script always crashes on all Unix machines when runs long enough. It seems there is memory leak.

### Steps to reproduce the behavior
typical Isis reduction script demonstrated this behaviour are located at:
 [https://github.com/mantidproject/scriptrepository/blob/master/direct_inelastic/LET/ReductionWrapper_withPerformance.py](url) -- logger
and:
[https://github.com/mantidproject/scriptrepository/blob/master/direct_inelastic/MERLIN/MERLINReduction_2018_2WithPerformance.py](url)
An executor script. Other executors are availible but this one is the most efficient one. 
The issue occurs the faster the lower memory a machine has, though isiscompute with 0.5Tb of memory is also susseptable when mutliple users occupy substantial chunk of memory. 

To demonstrate the issue, one should place these two files in a single folder of a unix machine which has access to archive mounted at root. The mahcine should also have isis instrument parameters in /usr/local/mprogs/InstrumentFiles. See the executor script rows 150-151 where necessary path-es to the data are specified.  The results and the log files are written into the script folder. 
The typical issue can be best reproduced on 32Gb machine as follows. 
The reduction script starts and have the following memory and performance characteristics, printed in the log file:

*** Test started:  29/10/2018 at 17:39
*** Self Memory: 487.6640625Kb; Kids memory 136.0234375Kb
***                    total        used        free      shared  buff/cache   available
***      Mem:          32169         700       30789          13         679       31027
***      Swap:          2047         647        1400
*** ---> File MER41624.nxs processed in 76.16sec
*** Self Memory: 15167.0Kb; Kids memory 305.0Kb
***      Mem:          32169       15322       16087          13         758       16401
***      Swap:          2047         647        1400
*** ---> File MER41625.nxs processed in 33.68sec
*** Self Memory: 18495.0Kb; Kids memory 14860.0Kb
***      Can not launch subprocess to evaluate free memory.
*** ---> File MER41626.nxs processed in 34.54sec
*** Self Memory: 18495.0Kb; Kids memory 14860.0Kb
***      Mem:          32169        5221       26103          13         844       26502
***      Swap:          2047         647        1400 

The memory, reported by resources module starts creeping up. 
When this memory reaches ~19Mb the script can not start subprocess to evaluate memory any more:

*** ---> File MER41650.nxs processed in 32.15sec
*** Self Memory: 18495.0Kb; Kids memory 14860.0Kb
***      Can not launch subprocess to evaluate free memory.
*** ---> File MER41651.nxs processed in 32.60sec
*** Self Memory: 18495.0Kb; Kids memory 14860.0Kb
***      Can not launch subprocess to evaluate free memory.
*** ---> File MER41652.nxs processed in 31.37sec
*** Self Memory: 18495.0Kb; Kids memory 14860.0Kb 

When this memory reaches ~25Mb Mantid crashes.

*** ---> File MER41842.nxs processed in 63.45sec
*** Self Memory: 24362.0Kb; Kids memory 14860.0Kb
***      Can not launch subprocess to evaluate free memory.
*** ---> File MER41843.nxs processed in 30.09sec
*** Self Memory: 24564.0Kb; Kids memory 14860.0Kb
***      Can not launch subprocess to evaluate free memory.
*** ---> File MER41844.nxs processed in 38.35sec
*** Self Memory: 24625.0Kb; Kids memory 14860.0Kb
***      Can not launch subprocess to evaluate free memory.
Crash 

The critical values differ sligtly on different memory machines but the general trent is the same. 
Note. When isiscompute is not occupied, the creep-up does not occur. 
When it has a lot of users, the behaviour is similar. 

The issue is critical for excitation group. 

### Platforms affected
Unix (RHEL7 though nobody tried it on Windows)

martyngigg, 2018-10-30:
@abuts How are the above scripts being executed. Are left to run in `MantidPlot` continuously?

abuts, 2018-10-30:
@martyngigg  Its in script repostiory, LET section.

martyngigg, 2018-12-08:
It was found that a major culprit for this was that the results log in MantidPlot retains everything that is ever written to it. In particular in debug setting this causes huge memory drain after only a couple of hours running.

closed 2021-03-01 by stale[bot]