mantidproject/mantid #23922: Workbench Font scaling issues when remoting from PC on 4k to PC on 1080p

DTasev, 2018-11-02:
### Expected behavior
Fonts should be scaled correctly.

### Actual behavior
Picture of the issue:
[image]

As can be seen, some fonts are scaled, some (text on buttons) are not. This does not seem to be Workbench specific issue, as it also happens on CMake (also Qt based GUI).

Hopefuly there is a Qt configuration to select or manually set the font scaling.

### Steps to reproduce the behavior
Current crude way of reproducing:
- Log into a PC with 4k resolution
- Remote into another PC with 1080p
- Launch the workbench

### Platforms affected
Only Windows has been tested.

martyngigg, 2019-05-02:
This has also been observed by a user. They had a remote desktop session from a laptop to their desktop and when logging into the local machine again the workbench display was similar to the image above. Both laptop and desktop monitor had a resolution of 2560x1440.

martyngigg, 2019-06-19:
This will prevent adoption of workbench for the above user.

martyngigg, 2019-06-19:
On the user setup the script editor also has mangled text where it looks very much like the picture in #24520.

NickDraper, 2019-07-05:
Harry, could you find some time to look into this even if it means putting down other release tasks for a day or two.

hsautessella, 2019-07-05:
Yep, no problem 👍

closed 2019-07-10 by NickDraper