mantidproject/mantid #24009: SegFault in python.Muon unit tests with python3

nvaytet, 2018-11-14:
### Expected behavior

Running `ctest -R python.Muon` should not produce any errors.

### Actual behavior

`ctest -R python.Muon` has 4 failed tests via `SegFault` on Ubuntu 18.04 with Python3:
```
The following tests FAILED:
	2133 - python.Muon.AxisChangerView_test.AxisChangerView_test (SEGFAULT)
	2137 - python.Muon.LoadWidgetPresenter_test.LoadWidgetPresenter_test (SEGFAULT)
	2138 - python.Muon.LoadWidgetView_test.LoadWidgetView_test (SEGFAULT)
	2142 - python.Muon.PeriodicTablePresenter_test.PeriodicTablePresenter_test (SEGFAULT)
```
The same with Python2 produces no errors.

The last lines of a log from a debug run:
```
2132: GeneratePeaks-[Debug] Peak function 15: Gaussian (Height, PeakCentre, Sigma)
2132: GeneratePeaks-[Debug] Peak function 16: IkedaCarpenterPV
2132: GeneratePeaks-[Debug] Peak function 17: IkedaCarpenterPV (I, Alpha0, Alpha1, Beta0, Kappa, SigmaSquared, Gamma, X0)
2132: GeneratePeaks-[Debug] Peak function 18: Lorentzian
2132: GeneratePeaks-[Debug] Peak function 19: Lorentzian (Amplitude, PeakCentre, FWHM)
2132: GeneratePeaks-[Debug] Peak function 20: PseudoVoigt
2132: GeneratePeaks-[Debug] Peak function 21: PseudoVoigt (Mixing, Height, PeakCentre, FWHM)
2132: GeneratePeaks-[Debug] Peak function 22: Voigt
2132: GeneratePeaks-[Debug] Peak function 23: Voigt (LorentzAmp, LorentzPos, LorentzFWHM, GaussianFWHM)
2132: ConfigService-[Debug] Looking for  at ISIS.
2132: FacilityInfo-[Debug] Blank instrument specified, using default instrument of .
2132: ConfigService-[Debug] Looking for  at ISIS.
2132: FacilityInfo-[Debug] Blank instrument specified, using default instrument of .
2132: FileValidator-[Debug] Encountered exception while checking for writable: File not foundEncountered exception while checking for writable: File not foundEncountered exception while checking for writable: File not foundEncountered exception while checking for writable: File not foundEncountered exception while checking for writable: File not foundEncountered exception while checking for writable: File not foundDo not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: FileValidator-[Debug] Do not have enough information to validate "SCDCalibrate.DetCal"
2132: ConfigService-[Debug] Unable to find curvefitting.peakRadius in the properties file
2132: ConfigService-[Debug] Unable to find curvefitting.peakRadius in the properties file
2132: ConfigService-[Debug] Unable to find curvefitting.peakRadius in the properties file
1/1 Test #2132: python.Muon.AxisChangerView_test.AxisChangerView_test ...***Exception: SegFault  2.54 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   2.63 sec

The following tests FAILED:
    2132 - python.Muon.AxisChangerView_test.AxisChangerView_test (SEGFAULT)
Errors while running CTest
```

Also running `PYTHONPATH=$PWD/bin:/path/to/my/source/scripts gdb --args python3 /path/to/my/source/scripts/test/Muon/AxisChangerView_test.py` did not help:

```
Starting program: /usr/bin/python3 /home/nvaytet/work/code/mantid/sources/py3/scripts/test/Muon/AxisChangerView_test.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe55f0700 (LWP 11231)]
[New Thread 0x7fffe4def700 (LWP 11232)]
[New Thread 0x7fffdffff700 (LWP 11233)]
[New Thread 0x7fffdf7fe700 (LWP 11234)]
.
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007fffe9425831 in gtk_container_add () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
(gdb)
```

### Steps to reproduce the behavior

`ctest -R python.Muon` on Ubuntu 18.04 with Python3

### Platforms affected

Ubuntu 18.04 + Python3

AnthonyLim23, 2019-05-20:
does this error still occur?

nvaytet, 2019-05-20:
I no longer see these tests in the unit tests, so I guess the error no longer occurs, since the tests have been removed...

closed 2019-05-20 by AnthonyLim23