mantidproject/mantid #24031: FileFinder and OnCat slow for lots of files

rosswhitfield, 2018-11-16:
On WAND² they often load ~1800 files at once, (180 degrees in 0.1 degree steps) and using FileFinder it's slow to get the list of files.

`FileFinder.findRuns('WAND27976-29776')` takes > 1 minute to return. OnCat has the potential to be so much faster! Getting `https://oncat.ornl.gov/api/datafiles/?facility=HFIR&instrument=HB2C&projection=location&ranges_q=indexed.run_number:27976-29776` takes about 0.1 seconds.

@PeterParker

PeterParker, 2018-11-17:
I agree we have the potential to do much better with file-finding in general, but wanted to keep the recently-merged [ONCat PR](https://github.com/mantidproject/mantid/pull/23879) limited in scope so this issue was never going to be addressed as part of that work.

To solve this issue we'd need at least a partial re-write of `FileFinder`, as well as changes to `IArchiveSearch`.  This would mean we could move away from multiple under-the-hood calls to `FindRun` towards an approach where we'd be able to find a range of files all at once.  My concern here, though, is that so much rests on top of file-finding and it would be so easy to break some edge-case functionality that is not directly under test.

I think there's general agreement that [`FileFinder` needs to be reworked, probably from the ground up](https://github.com/mantidproject/mantid/issues/8360), but this would be a large change that in my opinion would be better suited for a developer still on the project.

I'd like to bring this up for discussion in one of the US Mantid standup meetings, but unfortunately I won't be able to attend on Monday.

PeterParker, 2018-11-19:
... and as a temporary workaround, of course, you could just do something like this:

```python
import requests

ranges = '27976-29776'

URL = (
    'https://oncat.ornl.gov/api/datafiles/'
    '?facility=HFIR'
    '&instrument=HB2C'
    '&projection=location'
    '&tags=type/raw'
    '&exts=.nxs.h5'
    '&ranges_q=indexed.run_number:' + ranges
)

datafiles = requests.get(URL).json()
locations = [datafile['location'] for datafile in datafiles]

print(locations[0])
```

This would print:

```
/HFIR/HB2C/IPTS-7776/nexus/HB2C_27976.nxs.h5
```

peterfpeterson, 2021-08-30:
WIP

closed 2022-03-05 by stale[bot]