mantidproject/mantid #24075: Refactor `AlignAndFocusPowderFromFiles` to save and load summed cache files

marshallmcdonnell, 2018-11-21:
**Original reporter:** @marshallmcdonnell 

### Expected behavior
If we pass in a list of runs into `AlignAndFocusPowderFromFiles`, we create a cached file for the summed runs, not just individual cache files per run. Also, have the mechanism in-place to find either 1) the summed cache file for all listed runs, 2) a summed cache file for a subset of runs listed, or 3) individual cache files for the runs, in this order.

### Actual behavior
We save and load individual cache files. This is slow (at least for `NOMAD` cache files) for summing many runs.

The loop to refactor is found starting here: https://github.com/mantidproject/mantid/blob/fdce3b20d7d843e44ffca31d25daea62bd71de2b/Framework/PythonInterface/plugins/algorithms/AlignAndFocusPowderFromFiles.py#L304

### Steps to reproduce the behavior
This is an enhancement / new feature
1. Run  `AlignAndFocusPowderFromFiles` once for two or more runs
2. Rerun  `AlignAndFocusPowderFromFiles` to ensure the summed cache file of both files is loaded versus the two individual runs and then summed inside  `AlignAndFocusPowderFromFiles`.

### Platforms affected
N/A

marshallmcdonnell, 2019-01-30:
This script can be used to test the performance:

```python
import os
from mantid.simpleapi import *
import time
import argparse
import contextlib

parser = argparse.ArgumentParser(description='Performance for summing in AlignAndFocusPowderFromFiles.')
parser.add_argument('--cleanup', action='store_true', default=False,
                    help='Cleanup the cached files')
parser.add_argument('--recreate', action='store_true', default=False,
                    help='If running cleanup, recreate the cache files')

args = parser.parse_args()

@contextlib.contextmanager
def stopwatch():
    '''Wrap algorithm in this context manager to time how long they take.'''
    start = time.time()
    yield
    end = time.time()
    total = end - start
    print('Algorithm took %f seconds.' % total)
    print('')

# setup args
outname="output_wksp"
filenames = ['/SNS/NOM/IPTS-17210/nexus/NOM_77521.nxs.h5',
             '/SNS/NOM/IPTS-17210/nexus/NOM_77532.nxs.h5',
             '/SNS/NOM/IPTS-17210/nexus/NOM_77543.nxs.h5']

alignAndFocusArgs = dict(Filename=','.join(filenames),
                         GroupingWorkspace='grp_wksp',
                         MaxChunkSize=8,
                         CalFileName='/SNS/NOM/shared/CALIBRATION/2016_2_1B_CAL/NOM_d77540_2016_07_23_shifter.h5',
                         ResampleX=-6000,
                         TMin=300.0, TMax=16667.0,
                         CacheDir='/tmp')

CreateGroupingWorkspace(InstrumentName="NOMAD", GroupDetectorsBy='All', OutputWorkspace='grp_wksp')

# cleanup
if args.cleanup:
    files = [os.path.join('/tmp', filename) for filename in os.listdir('/tmp')
             if filename.startswith('NOM') and filename.endswith('nxs')]
    for filename in files:
        os.remove(filename)

    if args.recreate:
        # initially with all grouping
        AlignAndFocusPowderFromFiles(OutputWorkspace=outname,
                                     **alignAndFocusArgs)
        NormaliseByCurrent(InputWorkspace=outname, OutputWorkspace=outname,
                           RecalculatePCharge=True)

with stopwatch():
    AlignAndFocusPowderFromFiles(OutputWorkspace=outname,
                                 **alignAndFocusArgs)
    NormaliseByCurrent(InputWorkspace=outname, OutputWorkspace=outname,
                       RecalculatePCharge=True)
```

From local testing, here are results:
 - Initially (no caching): `Algorithm took 404.192959 seconds.`
 - After caching: `Algorithm took 50.963488 seconds.`

And what would be projected:
 - Ideal after adding summed cache files: `Algorithm took 50.963488 seconds.`17.0 seconds`

yxqd, 2019-01-30:
This first implementation took 13seconds to run with the summed cache

yxqd, 2019-01-30:
Next step: support partial sum.

Support arbitrary partial sum would be difficult. The number of combinations of possible partial sums grow exponentially with N, the number of files. Therefore we will sort the files, and then only care about partial sums that are contiguous. There will be `~N^2` possible partial sums that way.

PR #24653 (2019-02-04): Refactor `AlignAndFocusPowderFromFiles` to save and load summed cache 
**Description of work.**

**Report to:** 
@marshallmcdonnell  

**To test:**

Fixes #24075 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolation?
- If there are changes in the release notes then do they describe the ch [... cut]
closed 2019-02-04 by peterfpeterson