mantidproject/mantid #24114: Workspace/Algorithm Smoke Test

keeeto, 2018-11-27:
# Workspace/algorithm smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* Load some workspaces, try 5 from different instruments
* Show detectors, check detector view GUI, follow non-Python directions [in the documentation](https://www.mantidproject.org/MantidPlot:_Instrument_View)
	* Render tab [ ]
	* Draw tab [ ]
	* Pick tab [ ]
	* Instrument tree tab [ ]
* Test 5 - 10 random algorithms, follow [the documentation](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms)
	* Help button [ ]
	* Validation of inputs [ ]
	* Overwriting workspace [ ]
	* Run the algorithm [ ]
* Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus). [ ]

gvardany, 2018-12-07:
Tested nightly on macOS 10.11 with no other instance of mantid. Seems alright.

DanielMurphy22, 2019-11-12:
Tested on MacOS Mojave - So far only WB and first half of instructions

Instrument view on Mac Workbench Shows All cylindrical and Spherical views in a quarter of the window pane Issue #27244 

Some of the Peak button functionality of InstrumentView Doesn't seem to work, such as these buttons: 
[image]
Applying detector masks in the draw tab doesn't seem to work.

RichardWaiteSTFC, 2019-11-13:
Tested MantidPlot and workbench on windows 10. There are some issues: 
(1) On workbench, selecting a workspace before executing an algortihm doesn't default to have the input workspace same as selected (as it does in MantidPlot)

[image]

(2) It is possible to add peaks to a PeaksWorkspace from different instruments when the existing peaks from one instrument in that workspace are associated with a detectorID present in the other workspace. Is it possible to throw a warning - as it doesn't seem to correctly calculate that d-spacing/Q etc. In the screenshot below the highlighted row corresponds to the same detector and d-spacing (5 Ang) peak in MARI (which admittedly wouldn't be used for single-crystals) as the peak in the bottom table. The only difference is that some SXD peaks already existed in the top table.

[image]

it should be possible to throw a warning similar to when the tables are combined using CombinePeaksWorkspace 

[image]

RichardWaiteSTFC, 2019-11-13:
Have raised these as issues #27362 and #27365 respectively

RichardWaiteSTFC, 2019-11-13:
Tested on IDaaS. ran into an error on both workbench and mantidplot when trying to NormaliseByMonitor SXD data (for which there was no monitor). It did throw an error message, but it also caused a crash.

[image]

otherwise no issue

DanielMurphy22, 2019-11-13:
NormaliseToMonitor has an issue created #27021

ConorMFinn, 2019-11-13:
Tested on Ubuntu 18.04

No issues found in plot or workbench.

closed 2019-11-14 by DanielMurphy22