mantidproject/mantid #24115: Python Smoke Tests

keeeto, 2018-11-27:
# Python smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* Test that the Python scripting window works, [directions here](http://docs.mantidproject.org/v3.7.2/interfaces/ScriptingWindow.html)
	* Editor options [ ]
	* Execution options [ ]
	* Script output [ ]	
* Perform some workspace algebra [ ]
* Test numpy functionality [ ]
* Use the scripting window to run some scripts [ ]
* Run through some examples from [the documentation](http://www.mantidproject.org/Python_In_Mantid), 3 or 4 examples is enough [ ]

gvardany, 2018-12-07:
Tested nightly on macOS 10.11 with no other instance of mantid. Seems alright.

keeeto, 2018-12-07:
Thanks @gvardany !

DanielMurphy22, 2019-11-12:
Tested on MacOS Mojave, Plot and WB. Instruction don't apply so perfectly to WB.

WB doesn't have functions getInstrumentView() or plotSpectrum() but I assume this is correct.

PhilColebrooke, 2019-11-13:
No problems on Windows 10 with MantidPlot or Workbench.

This is unimportant for this release, but in Plot there is a keyboard shortcut to close the current tab in the script editor (Ctrl+W) so should this be added to Workbench?

giovannidisiena, 2019-11-13:
Tested on Ubuntu 18.04, no issues found.

RichardWaiteSTFC, 2019-11-13:
No problems with MantidPlot or workbench on IDaaS

closed 2019-11-13 by DanielMurphy22