mantidproject/mantid #24116: Plotting Smoke Tests

keeeto, 2018-11-27:
# Plotting smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
	* Simple plot [ ]
	* Another way to plot [ ]
	* Adding curves to existing plots [ ]
* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
	* Plot all spectra [ ]
	* Plot from matrix [ ]
	* Change colour map [ ]
	* Contour plot [ ]
	* Spectrum viewer [ ]
* 3D plotting:
	* Load some data eg `LOQ7041` from the ISIS sample data
	* Double click the workspace to open the table view
	* Select the table
	* 3D plot becomes available as an option in the toolbar [ ]
	* 3D wire frame [ ]
	* 3D hidden line [ ]
	* 3D polygon [ ]
	* 3D wire surface [ ]

DanielMurphy22, 2019-11-12:
Tested on MacOS Mojave Plot and WB(Although obviously limited by what plotting WB can do!)

PhilColebrooke, 2019-11-12:
Tested on Windows 10, both MantidPlot and Workbench. Only problem I found was being unable to open figure options and the fitting menu after plotting spectra from a workspace's data table. This has now been fixed (#27356). Everything else works okay.

DannyHindson, 2019-11-13:
Tested on IDAaaS. Builds on there are from last Fri (8th Nov) so not fully up to date

MantidPlot all fine. Get following crash on Workbench when doing the 1D plot where you generate the plot by right clicking on rows in the matrix window. The problem happens when you open the "Configure Plot Options" button on the resulting plot window:

[image]

Don't get this when doing the same actions on the Windows nightly build from last night - so not sure if it's fixed or whether it's not visible on Windows

ConorMFinn, 2019-11-13:
Tested on Ubuntu 18.04

No issues observed on plot or workbench.

closed 2019-11-13 by DanielMurphy22