mantidproject/mantid #24117: Interface Smoke Tests

keeeto, 2018-11-27:
# Interface/misc smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* Mantid opens without errors or warnings [ ]
* Every option in `Interface` opens a GUI [ ]
* Load some test data [ ]
* Catalogue opens  [directions here](https://www.mantidproject.org/Catalog_Search) [ ]
* Saving/loading projects works  [directions here](https://www.mantidproject.org/MantidPlot:_The_File_Menu#File-.3E_Save_Project_as...) [ ]

giovannidisiena, 2019-11-11:
Tested against Windows 10. Mantid opens without errors or warnings and every option in Interface opens a GUI, however the contents of #27160 do not seem to have any effect.

martyngigg, 2019-11-11:
@giovannidisiena According to the git history that commit is in `release-next`. Looking at the code also confirms that: [mainwindow.py](https://github.com/mantidproject/mantid/blob/release-next/qt/applications/workbench/workbench/app/mainwindow.py#L377)

giovannidisiena, 2019-11-11:
> @giovannidisiena According to the git history that commit is in `release-next`. Looking at the code also confirms that: [mainwindow.py](https://github.com/mantidproject/mantid/blob/release-next/qt/applications/workbench/workbench/app/mainwindow.py#L377)

That's strange. Do you have any ideas why it might no longer be working?

NickDraper, 2019-11-11:
Started Tesing on iDAaaS

DanielMurphy22, 2019-11-11:
Tested on Mac, Mojave at ISIS. Currently just Plot.
- ISIS Reflectometry Old interface fails to load (@gemmaguest informed)
- Almost all interfaces are persistently on top! 👎 
     - Open 'Manage User Directories' and it appears behind the already open interface
     - In DynamicPDF interface click 'Load', which opens a "DynamicPDF Slice Selector" window, which 
       on Mac appears behind the original interface!!! So its easy to miss it!
- Elemental Analysis doesn't seem to be on top persistently! 👍 
- Indirect> Data Reduction, Tools and ISIS Reflectometry slow to load (3-5 seconds)
- Cosmetic: Direct>MSlice interface>Workspace Manager Tab:
                             - >MDHisto "Save to MantidPlot" button the wrong size
                             - >2D "Compose" drop-down menu (for scale and boss) looks bad!!!

alicerussell1, 2019-11-13:
Tested on Ubuntu 18 VM no issues on workbench or plot

closed 2019-11-14 by DanielMurphy22