mantidproject/mantid #24118: Fitting Smoke Tests

keeeto, 2018-11-27:
# Fitting smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

The check boxes should correspond to headings in the documentation (linked). Just follow the corresponding directions.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
	* Simple fit [ ]
	* Activating the fitting tools [ ]
	* Selecting the fitting range [ ]
	* Other Fit Function Settings properties [ ]
	* Setting up a fit model [ ]
	* Adjusting fit function parameter [ ]
	* Saving a model [ ]
	* Tying and constraining fit parameters [ ]
	* Execute your fit [ ]
	* Inspecting a fit result [ ]
* Additional options [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_fit_model_choices.html#fit-model-choices)
	* Non peak model + background [ ]
	* User defined function [ ]

alicerussell1, 2019-11-11:
Tested on windows 10, no issues found on workbench or plot.

DanielMurphy22, 2019-11-12:
Tested on macOS Mojave, Plot and WB and all seems fine.

RichardWaiteSTFC, 2019-11-12:
Tested on IDaaS (nightly build 8-Nov), no issues found ion Plot or Workbench

giovannidisiena, 2019-11-13:
Tested on Ubuntu 18.04, no issues found apart from duplication of cost functions and minimizers in their respective drop downs.

closed 2019-11-13 by DanielMurphy22