mantidproject/mantid #24119: Clean/Dirty Install Smoke Test

keeeto, 2018-11-27:
# Clean/dirty install smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

### Dirty install

* Make sure that you have several versions of Mantid installed
	* Last release
	* A nightly
	* If possible an old release
* Install the latest version of the new Mantid
* Check that it boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files
* Windows:
	* Uninstall the program
	* Clear shortcuts from desktop
	* Clean out the registry
		* Load regedit (Command Prompt > regedit)
		* Find `HKEY_CURRENT_USER > SOFTWARE > Mantid` delete it
* OSX:
	* Remove the application
	* Remove the `~/.mantid directory`
	* Remove `~/Library/Preferences/org.mantidproject.MantidPlot.plist`
* Linux
	* Remove the package: `/opt/Mantid`
	* Remove `~/.config/Mantid`
	* Remove `~/.mantid/`

* Re-install the latest version of the new Mantid
* Check that it boots up correctly

DTasev, 2018-12-06:
I'll do the install smoke testing on Windows and Ubuntu 16.04

keeeto, 2018-12-06:
Thanks @DTasev !!

DavidFair, 2019-11-11:
I'll pick this up for this release

Windows / Ubuntu / RHEL7 (all completely clean and dirty)

DavidFair, 2019-11-12:
This works for the above platforms on release 4.2

DanielMurphy22, 2019-11-13:
Testing on MacOS Mojave. On Mac, Mantid does NOT save a nightly as a separate application. Any reinstall just overrides the previous.

closed 2019-11-13 by DanielMurphy22