mantidproject/mantid #24150: Allow ISIS Powder to predict PDF vs Rietveld mode for POLARIS

ElliotAOram, 2018-11-30:
**Original reporter:** [Ron Smith and Helen Playford - ISIS]

POLARIS using the ISIS_Powder scripts currently runs in either `Rietveld` or `PDF` mode. This is currently manually defined by the user as a script parameter. Providing the log file is available, this can be calculated automatically. 
If the `frequency_block` in the logs is `50Hz` then it is Rietveld mode. `0Hz` is `PDF. The chopper shouldn't ever be anything different from these values.
In the following cases POLARIS instrument scientists would like the follow action: 

| Frequency block |  Log file available |  mode explicitly specified |  action  |          
| ------------------ | ------------------- | ---------------------------- | -------- |
| 0Hz                     |        Yes                |                   Yes                 | Use the mode that is specified by the user |
| 0Hz                     |        Yes                |                  No                  | Automatically use PDF (inform user with warning in log) | 
| not 0Hz or 50Hz |        Yes                |                 Yes                   | Use the mode that is specified by the user |
| not 0Hz or 50Hz |        Yes                |                 No                   | Raise a error stating that `please specify a mode of operation either 'PDF' or 'Rietveld'.` |
|      N/A               |         No                |                  Yes                  | Use the mode that is specified by the user |
|      N/A               |         No                |                  No                  | Raise a error stating that `please specify a mode of operation either 'PDF' or 'Rietveld'.` |
| 50Hz                     |        Yes                |                   Yes                 | Use the mode that is specified by the user |
| 50Hz                     |        Yes                |                  No                  | Automatically use Rietveld (inform user with warning in log) |

ElliotAOram, 2019-03-07:
@SamJenkins1 can you look into this after release?

SamJenkins1, 2019-03-07:
ok

ElliotAOram, 2019-03-07:
Thanks, I've pushed all the work I have done so far with this - might be helpful

closed 2019-04-30 by martyngigg