mantidproject/mantid #24166: Unscripted Testing ISIS Indirect interfaces

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow the directions at: http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

DTasev, 2018-12-04:
Very easy instructions to follow! No issues found, apart from a few places where the docs are slightly out of date.

[Data Reduction](http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html#data-reduction)
Part 2, step 24 should say E-Low, E-Width and E-High, I believe
[Data Analysis Elwin](http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html#data-analysis-elwin)
LaTex output isn't rendered correctly
[Data analysis I(Q, t)](http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html#data-analysis-i-q-t)
6 - the `_iqt` workspace is 87x6, not 87x51
8 - No button called `Plot Result`, probably renamed to `Plot Spectrum`.
Nothing about `Tiled Plot`
[Data analysis I(Q, t) Fit](http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html#data-analysis-i-q-t-fit)
9 - Can't change range to 0 - 10 as the workspace from last part was 87x6. Works okay for spectra 0-5 with the workspace I had
12 - `Plot Result` button doesn't exist

keeeto, 2018-12-05:
Thanks @DTasev

closed 2018-12-05 by keeeto