mantidproject/mantid #24167: Unscripted Testing Live Data ORNL

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

mdoucet, 2018-12-12:
I've been testing the live data on MR throughout November and December and it still works great (apart from a minor issue that affected my instrument and that has since been resolved).

closed 2018-12-12 by mdoucet

keeeto, 2018-12-12:
Thanks @mdoucet !