mantidproject/mantid #24173: Unscripted Testing Mantidplot Group 2

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://www.mantidproject.org/Unscripted_Manual_Testing#Groups

keeeto, 2018-12-05:
Hi @aybamidele are you okay to do this testing?

aybamidele, 2018-12-05:
Yes, I am currently working on it. I have found a few problems and have created issues for them. I should mention that I am not aware of exactly what to do, but I've asked a few people, the idea is to try to break stuff in group 2 (which is the portion I have been assigned)

keeeto, 2018-12-05:
Yep that is the idea. Thanks @aybamidele . Also, I will take that on board and update the instructions. Cheers.

keeeto, 2018-12-05:
Hi @aybamidele - are you done with this issue? Feel free to close it if finished. Cheers.

closed 2018-12-05 by aybamidele

keeeto, 2018-12-06:
Thanks @aybamidele