mantidproject/mantid #24174: Unscripted Testing Mantidplot Group 1

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://www.mantidproject.org/Unscripted_Manual_Testing#Groups

keeeto, 2018-12-05:
Hi @robertapplin are you okay to do this testing?

robertapplin, 2018-12-05:
Hi @keeeto. Yes, I have started it and I am about half way through but haven't found any errors yet

keeeto, 2018-12-05:
Thanks @robertapplin - just checking in.

keeeto, 2018-12-05:
Thanks @robertapplin