mantidproject/mantid #24181: Unscripted Testing Multi dataset fitting

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: https://www.mantidproject.org/Fitting_QENS_I(Q,_t)

DTasev, 2018-12-03:
Nightly Used: mantidnightly-3.13.20181130.1602-win64
OS: Windows 10

# Critical - causes Mantid to crash
Steps 17 or 18 (whichever is done first, the order for these 2 doesn't matter) from [the instructions](https://www.mantidproject.org/Fitting_QENS_I(Q,_t)) hard crash Mantid. This seems to only happen after the tie from step 13 is added.

<details>
<summary>Steps 17, 18 body</summary>
17. Double click on FlatBackground.A0 and a new window will open. Enter a value into the Value box of the first spectrum and in Set select Set to all.
<br>18. Repeat for StretchExp.Lifetime.
</details>

# Important
## Can't run the Fit
To prevent the crash described above - set values of all spectra, and then add the tie. Trying to run the fit shows the following error window
[image]

## Changing A0 value after tie
Do all steps until 16 - enter a value into A0 (without clicking the `...` to open the dialogue), and press `Enter`: it shows the error reporter.

## Documentation updates
### On [testing page](http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html#data-analysis-i-q-t)
7. Click `Plot Current View` this should plot the same data as the preview window
8. Click `Plot Result` this should give a plot with the title irs26176_graphite002_iqt

These 2 need to be updated - neither of the buttons exists anymore in the GUI.

### On [fitting page](https://www.mantidproject.org/Fitting_QENS_I(Q,_t))
#### Tie spelling mistake
Step `13` has a spelling mistake in the tie-in string:

> Enter 1.0-FlatBackgound.A0 & a box showing the Tie will appear.

`1.0-FlatBackgound.A0` - the work `Backgound` is missing an R and should be `Background`. This mistake makes it an invalid tie.

Additionally, the `1.0-FlatBackground.A0` instruction should be in `code`, or **bold** to visually separate it from the rest of the sentence.

#### Out-of-date documentation
Step 17 says to double click `A0` which doesn't seem to do anything anymore: I think the functionality has been moved to the `...` button that is now showed when the field is focused - the docs should be updated to reflect that change

Step 19 talks about a `Output Property` which now seems to be called `OutputWorkspace` - consider changing the docs to reflect that

There is a mention of a bug as of July 2015 - check if the bug still persists and update accordingly (if fixed - remove, if present - update date)

keeeto, 2018-12-04:
@DTasev thanks for raising these issues! Other than these, are you finished this testing?

DTasev, 2018-12-04:
@keeeto I have finished the testing. I am happy to try it out again when the issues have been addressed

keeeto, 2018-12-04:
Thanks Dimitar! I'll close this issue for now, just to keep on top of what has/has not been done.

closed 2018-12-04 by keeeto