mantidproject/mantid #24183: Unscripted Testing DynamicPDF

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

jmborr, 2018-12-06:
Tested on installed version of `mantidnightly`
- [x] load ARCS reduced data 
- [x] display reduced data
- [x] fit of data with built-in fit model
- [x] Fourier transform of the residuals with custom settings

closed 2018-12-06 by jmborr

keeeto, 2018-12-06:
Thanks @jmborr