mantidproject/mantid #24185: Unscripted Testing Engineering Diffraction

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: http://docs.mantidproject.org/nightly/interfaces/Engineering%20Diffraction%20Test%20Guide.html

AnthonyLim23, 2018-12-03:
found a bug that means it does not work with data archive #24197

AnthonyLim23, 2018-12-04:
going to just list issues here then create an issue later:

1. If the calibration run number is not a calibration run then the failed message appears in the logger. It would be nice to have a pop up with the message. It is annoying to have to scroll through the logger for auto-saving to reset the logger's view.

2. The current calibration box looks like it has line edits when it does not. 

3. I could not get the cropped calibration to work with spectrum numbers (this might be that I put bad values in).

4. using browse to load a non Enginex (I used muon data) into the calibration sample number caused an unhelpful error message about the vanadium being wrong. After changing the calibration sample back to a valid value I had to replace the vanadium value to get it to work again. 

5. the connection to the data archive seems buggy I just got the message below. I cannot close mantid then without causing it to crash. I can no longer load the data.
```
ISISDataArchive::getPath() - fName=ENGINX00062260.txt
Sending http GET request to: http://data.isis.rl.ac.uk/where.py/windir?name=ENGINX00062260.txt
Could not access archive index InternetError: Connection and request failed TimeoutISISDataArchive::getPath() - fName=ENGINX00062260.csv
Sending http GET request to: http://data.isis.rl.ac.uk/where.py/windir?name=ENGINX00062260.csv
```

keeeto, 2018-12-05:
@Pasarus , @AnthonyLim23 seems to have broken his computer to the point that it can't even run the interface anymore. If you manage the same then it's safe to say this interface is a total basket case :)

Pasarus, 2018-12-05:
@keeeto It says in the test document that a box should pop up the first time this is ran and ask for an RB number? This has not happened for me, is that an issue with the GUI or the test itself?

AnthonyLim23, 2018-12-05:
its in the top corner of the GUI

Pasarus, 2018-12-05:
Just spoke to Sam Jackson and he said that the pop up box has actually been removed because it was "Annoying" so the test is just out of date.

Pasarus, 2018-12-05:
So I have the same issue as @AnthonyLim23 now, I am going to wait and do a little investigating. See if it fixes itself.

Pasarus, 2018-12-05:
After some coaxing and turning off the data search archive I managed to get it to work for the most part. Nothing that was reproducible occurred.

closed 2018-12-05 by Pasarus