mantidproject/mantid #24189: Unscripted Testing Sample Transmission calculator

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See http://docs.mantidproject.org/v3.7.2/interfaces/SampleTransmissionCalculator.html

aybamidele, 2018-12-04:
@keeeto I'm not sure who handles this interface, but I did notice that it is possible in the multiple input wavelength range, you can start from negative values of wavelength and the code still works.

keeeto, 2018-12-04:
Good spot @aybamidele - can you create an issue related to this and assign it to me. I'll look into assigning it to a developer. I also don't know who does this area at the moment!!

keeeto, 2018-12-04:
Also, once you're done testing, just close this issue.

closed 2018-12-04 by aybamidele