mantidproject/mantid #24191: Unscripted Testing ISIS SANS (v2 GUI)

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

# SANS GUI Testing

<div class="contents" data-local="">

</div>

## Data reduction

*Preparation*

  - Get the ISIS sample data from the website
  - Ensure that the ISIS sample data directory is in the search
    directories

**Time required 20 - 30 minutes**

-----

1.  Open `Interfaces` \> `SANS` \> `SANS v2`

2.  Set `Instrument` to `LOQ`

3.  Choose `Load User File`; from the ISIS sample data in the `loqdemo`
    folder, choose `Maskfile.txt`

4.  Choose `Load Batch File`; from the ISIS sample data in the
    <span class="title-ref">loqdemo</span> folder, choose
    `batch_mode_reduction.csv`

5.    - In the `Settings` tab:
        
          - Go to `Mask`
          - Click `Display Mask`
          - This should give an instrument view with a circle at the
            centre

6.    - In the `Runs` tab
        
          - Click `Process`
          - After some seconds the rows should turn green
          - In the Main window there should be a series of new
            workspaces; 3 group workspaces and 4 2D workspaces
          - Change the first column of the first row to 74045; click
            process
          - The row should turn blue; hovering over the row should give
            an error message
          - Change the `Reduction` button to 2D
          - Check the `Plot Results` box
          - Click `Process`
          - A plot window will open; initially empty, then with a red
            line
          - Change `Reduction` back to 1D
          - Click `Process`
          - In the plot you should end up with a red and a black line
            plotted
          - Check the `Multi-period` box
          - 6 additional rows should appear in the table
          - Check that the `Add`, `Delete`, `Copy`, `Paste`, `Cut` and
            `Erase` icons work as expected
          - Delete all rows and re-load the batch file as previously

7.    - In the `Settings` tab
        
          - Set `Reduction Type` to `Merged`
          - Return to the `Runs` tab
          - Ensure `Plot results` is checked
          - Click `Process`
          - This should result in a plot with three lines

8.    - In the `Beam centre` tab
        
          - Click
          - A plot should appear after some seconds, with 4 lines
          - The 4 lines should gradually get closer together
          - This will run for some time, probably minutes

9.    - In the `Sum Runs` tab
        
          - Enter `74044, 74019` in the top line
          - Enter `LOQ74044-add` as Output file
          - Ensure that the `Output directory` is in you r managed paths
          - Click `Sum` at the bottom
          - Go back to the `Runs` tab
          - Remove all rows
          - Reload the batch file as before
          - Change the first column of both rows to `LOQ74044-add`
          - Click `Process`
          - This should now process as before

10.   - In the `Diagnostic Page` tab
        
          - For run choose `Browse` and load the `LOQ74044.nxs` file
          - Click each of the `Integral` buttons
          - They should produce plots
          - Check the `Apply Mask` boxes and click the buttons again
          - They should produce new, slightly different plots

11.   - In the `Runs` tab
        
          - Check all variables and inputs for tool tips by hovering
            over them

TTitcombe, 2018-12-03:
@robertapplin Note that as of PR #24128 step 2. is not longer applicable. The instrument should be determined by the user file, and processing is not possible until this is done

robertapplin, 2018-12-04:
@TTitcombe I am getting an error when I get to step 5 of the workflow above:
[image]

keeeto, 2018-12-04:
@robertapplin can you re-create this error, turn it into an issue and assign @TTitcombe - also add milestone 3.14 to it.

robertapplin, 2018-12-04:
@TTitcombe I am now getting the error:
[image]
when I get to step 5. Is the data file I am missing in the data archive?

keeeto, 2018-12-04:
@robertapplin that file should be in the ISIS sample data in the `loqdemo` directory. It might be necessary to add that directory to your path.

robertapplin, 2018-12-04:
@TTitcombe @keeeto  Ah yes, I had the sample data directory as a default directory, but it still couldn't find it because it was looking for a file with name `74044`, but the file had the name `LOQ74044`. Is this a possible problem?

keeeto, 2018-12-04:
@robertapplin  Ah probably worth checking that your instrument is set to LOQ.

robertapplin, 2018-12-04:
@keeeto yea, it seems the instrument has automatically been set to LOQ as it has been determined by the User File. I'm not too sure where to go from here

keeeto, 2018-12-04:
Try by putting in the full name. But if it is not working with just run number + instrument set to LOQ, then this is still a bug.

TTitcombe, 2018-12-04:
@robertapplin that's another bug which is approved and ready for merged. To fix locally:
On mantidPlot -> Help -> First time setup -> Set the default instrument to LOQ

keeeto, 2018-12-05:
Thanks @robertapplin

closed 2018-12-05 by keeeto