mantidproject/mantid #24195: Unscripted testing - project recovery

keeeto, 2018-12-03:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

keeeto, 2018-12-05:
Hi @samueljackson92 are you able to pick this up? Don't worry if you're overloaded, there are a few people around with capacity, just let me know.

samueljackson92, 2018-12-05:
Yup, as discussed just doing it now.

samueljackson92, 2018-12-05:
@keeeto Ran through the test cases. Seems to work ok even with my naive attempts to break it.

closed 2018-12-05 by samueljackson92