mantidproject/mantid #24223: SANS Python Testing fails on ImportErrors

TTitcombe, 2018-12-04:
Unit testing the Python SANS fails on master

### Expected behavior
Tests should pass

### Actual behavior
A lot of tests fail on `ImportError: No module named sans.foo.bar`
absolute_import is imported in the tests, but it still cannot find the relevant files

### Steps to reproduce the behavior
On master, run `ctest --output-on-failure -R PythonSANS`

### Platforms affected
At least ubuntu 16.04 and 18.04

TTitcombe, 2018-12-04:
@Mantid-Matthew

martyngigg, 2018-12-05:
So I think the problem here is a missing `site.py` file in the `bin` directory. It is currently only generated when doing a build with `ENABLE_WORKBENCH=ON` and it was not spotted that the file is not generated with this off.

closed 2021-03-01 by stale[bot]