mantidproject/mantid #24226: ORNL SANS - Allows negative number entry in Reduction Options

robertapplin, 2018-12-04:
### Actual behavior
In `ORNL SANS` -> `Reduction Options` tab you are allowed to enter negative numbers for Absolute scale factor,  Sample aperture diameter and Number of Q bins. I believe this shouldn't be allowed and could cause problems.

### Steps to reproduce the behavior
1.`Interfaces`->`SANS`->`ORNL SANS`->`Reduction Options` tab
2. Enter a negative number for any of the options stated above

### Platforms affected
All

mdoucet, 2018-12-12:
I'm not sure that the current status/use of this GUI is. Ricardo probably knows.

ricleal, 2018-12-12:
As far as know nobody at the ORNL uses the interfaces to reduce data.
The current ORNL SANS code in mantid has low priority in terms of maintenance.

closed 2021-03-02 by stale[bot]