mantidproject/mantid #24227: ORNL SANS - Crash when wrong file loaded and you try plotting

robertapplin, 2018-12-04:
### Expected behavior
No crash - validation of the loaded workspace before trying to plot.

### Steps to reproduce the behavior
1.`Interfaces`->`SANS`->`ORNL SANS`->`Detector` Tab
2. Tick `Use beam finder`
3. Load the file below
4. Click `Plot`
crash

**Data**
[irs26176_graphite002_red.zip](https://github.com/mantidproject/mantid/files/2644259/irs26176_graphite002_red.zip)

### Platforms affected
All

mdoucet, 2018-12-12:
I'm not sure that the current status/use of this GUI is. Ricardo probably knows.

ricleal, 2018-12-12:
As far as know nobody at the ORNL uses the interfaces to reduce data.
The current ORNL SANS code in mantid has low priority in terms of maintenance. Just what prevents users / scientists from work is fixed.

closed 2021-03-02 by stale[bot]