mantidproject/mantid #24230: SANS GUI Crashes Mantid

keeeto, 2018-12-04:
**Original reporter:** [username facility]/[nobody]. 

### Expected behavior

It should be possible to close the GUI without crashing Mantid.

### Actual behavior

If the GUI is closed before the initial loading has finished, Mantid crashes. This issue seems to have existed in previous versions of Mantid, therefore I suspect it is quite niche. It does, however, seem quite serious to me.

### Steps to reproduce the behavior

* Open Mantid
* Open Interfaces > SANS > SANS
* Quickly close the GUI (before the files have loaded up, a few seconds)
* Mantid crashes

### Platforms affected

OSX, Windows at least

TTitcombe, 2019-02-14:
As this is a bug with old SANS GUI which has not been reported by users, and that we are moving towards deprecation of this GUI, this is a low priority issue.

Mantid-Matthew, 2019-03-21:
As the old GUI is now deprecated I am closing this issue

closed 2019-03-21 by Mantid-Matthew