mantidproject/mantid #24253: Reflectometry GUI invalid options causes crash

ElliotAOram, 2018-12-05:
### Expected behavior
* The options box in the Reflectometry GUI should be validated or the expection thrown should be handled

### Actual behavior
* Adding invalid (unexpected) text to the `Options` field in the Reflectometry GUI causes a crash at processing stage

### Steps to reproduce the behavior
* Open Reflectometry GUI
* Load *.tbl
* Type `Test` into the `Options` field
* Click process
* Observe crash in Mantid

### Platforms affected
* Windows 10

gemmaguest, 2018-12-07:
This has existed for a while and not been noticed by users so I don't think it's high enough priority to hold up the release. It's not ideal but unless it's an easy fix I'd rather do it as part of the restructuring in #23630

gemmaguest, 2019-05-14:
This is fixed as part of the restructuring in #23051

closed 2019-05-14 by gemmaguest