mantidproject/mantid #24339: Segmentation fault when DownloadInstrument without internet connection

marshallmcdonnell, 2018-12-14:
**Original reporter:** @marshallmcdonnell 

### Expected behavior
If `DownloadInstrument` fails, it should do so "gracefully" and maybe with a better error than the all so helpful "Segmentation Fault"

### Actual behavior

This may be a special case but we are trying to run the Mantid Framework inside the nightly Docker containers [ [Dockerfiles](https://github.com/mantidproject/dockerfiles) and [Images on DockerHub](https://hub.docker.com/r/mantidproject/mantid/tags) ]. When issuing `import mantid.simpleapi`, the [`DownloadInstrument`](http://docs.mantidproject.org/v3.5.1/algorithms/DownloadInstrument-v1.html)  algorithm will cause a segmentation fault on starting up the Framework.  Will segmentation fault after writing the `github.json` in the `$HOME/.mantid` directory. 

The workaround for now is to issue the following bash command before trying to use the Framework (with the assumption you already have the instrument definition files):

```
echo "UpdateInstrumentDefinitions.OnStartup = 0" > /root/.mantid/Mantid.user.properties
```

With help of @quantumsteve, tried to get a backtrace using (pseudo-code) `gdb python && run && import mantid.simpleapi && bt` but running over night it never exited the import statement so never able to get to the `bt` command.

NOTE: Interestingly, running it a second time after the initial seg fault (running `python -c import mantid.simpleapi` again) will work without the seg fault and also won't call `DownloadInstrument` again.  I just assume this is because the  `$HOME/.mantid/github.json` exists makes the `DownloadInstrument` not run again. Example below of running command twice from an interactive session  launched using `docker run -it mantidproject/mantid:nightly_ubuntu16.04 /bin/bash -c "MANTIDPATH=/opt/mantidnightly/bin && export PYTHONPATH=${MANTIDPATH}:${PYTHONPATH} && bash"`

```
root@e9c4c6651ce0:/# python -c "import mantid.simpleapi"
/usr/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
FrameworkManager-[Notice] Welcome to Mantid 3.13.20181212.1445
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
DownloadInstrument-[Notice] Downloading 6 files from the instrument repository
Segmentation fault (core dumped)

root@e9c4c6651ce0:/# python -c "import mantid.simpleapi"
FrameworkManager-[Notice] Welcome to Mantid 3.13.20181212.1445
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
DownloadInstrument-[Notice] All instrument definitions up to date
```

### Steps to reproduce the behavior

Affects all three nightly docker images so pick either of the following (the `nightly` is equivalent to `nightly_centos7`):
```
docker run -it mantidproject/mantid:nightly_ubuntu16.04 /bin/bash -c "MANTIDPATH=/opt/mantidnightly/bin && export PYTHONPATH=${MANTIDPATH}:${PYTHONPATH} && python -c 'import mantid.simpleapi'"
```
or 
```
docker run -it mantidproject/mantid:nightly_centos7/bin/bash -c "MANTIDPATH=/opt/mantidnightly/bin && export PYTHONPATH=${MANTIDPATH}:${PYTHONPATH} && python -c 'import mantid.simpleapi'"
```

To get the logging at debug level, you can issue the following:
```
docker run -it mantidproject/mantid:nightly_ubuntu16.04 /bin/bash -c "mkdir /root/.mantid/ && echo 'logging.loggers.root.level=debug' > /root/.mantid/Mantid.user.properties && MANTIDPATH=/opt/mantidnightly/bin && export PYTHONPATH=${MANTIDPATH}:${PYTHONPATH} && python -c 'import mantid.simpleapi'"
```

I uploaded the output of this `debug` logging if it helps but I did not see anything immediately pop out at me:
[debug_output.txt](https://github.com/mantidproject/mantid/files/2679922/debug_output.txt)

### Platforms affected
Docker images of the nightly found [here](https://hub.docker.com/r/mantidproject/mantid/tags) but most likely all docker images or computers without internet connection, I assume.

One last note is we were trying to just use the docker containers for running tests with the Framework. Yet, issuing `pytest` would just issue the segmentation fault so it took some time investigating to pin point both the Mantid Framework and the `DownloadInstrument` algorithm.

peterfpeterson, 2018-12-17:
I've tried to reproduce this in a regular build by adding
```
CheckMantidVersion.OnStartup = 1
UpdateInstrumentDefinitions.OnStartup = 1
usagereports.enabled = 1
```
to my `~/.mantid/Mantid.user.properties` then `rm -rf ~/.mantid/instrument` before unplugging my network cable and starting `mantidplot`. I did not manage to reproduce the error that way.

peterfpeterson, 2018-12-17:
The important part of the log appears to be the last bit when processing a redirect from github:
```
InternetHelper-[Debug] Sending http POST request to: http://reports.mantidproject.org/api/feature
InternetHelper-[Debug] Answer from web: 302 Found
InternetHelper-[Debug] url relocated to https://github.com/mantidproject/mantid/raw/master/instrument/D11_Parameters.xmlSending https GET request to: https://github.com/mantidproject/mantid/raw/master/instrument/D11_Parameters.xml
InternetHelper-[Debug] Answer from web: 201 Created
InstrumentDataService-[Debug] PN6Mantid6Kernel11DataServiceINS_8Geometry10InstrumentEEE cleared.
AnalysisDataService-[Debug] PN6Mantid6Kernel11DataServiceINS_3API9WorkspaceEEE cleared.
PropertyManagerDataService-[Debug] PN6Mantid6Kernel11DataServiceINS0_15PropertyManagerEEE cleared.
InternetHelper-[Debug] Answer from web: 302 Found
InternetHelper-[Debug] url relocated to https://raw.githubusercontent.com/mantidproject/mantid/master/instrument/D11_Parameters.xmlSending https GET request to: https://raw.githubusercontent.com/mantidproject/mantid/master/instrument/D11_Parameters.xml
```

peterfpeterson, 2018-12-17:
Fixing the name of the logger in `GitHubApiHelper` you get a clearer picture of what is actually going on (even if I can't reproduce the error):
```
DownloadInstrument-[Notice] DownloadInstrument started
DownloadInstrument-[Information] Algorithm: DownloadInstrument v1
DownloadInstrument-[Information] Execution Date: 2018-12-17 17:11:59.698887000
DownloadInstrument-[Information] Execution Duration: -1 seconds
DownloadInstrument-[Information] UUID: 23a1bf32-3a6f-44f3-bd64-2b1087a85090
DownloadInstrument-[Information] Parameters:
DownloadInstrument-[Information]   Name: ForceUpdate, Value: 1, Default?: No, Direction: Input
DownloadInstrument-[Information]   Name: FileDownloadCount, Value: 0, Default?: Yes, Direction: Output
InternetHelper-[Debug] DownloadFile : https://api.github.com/repos/mantidproject/mantid/contents/instrument to file: /home/pf9/.mantid/instrument/github.json
InternetHelper-[Debug] Sending https GET request to: https://api.github.com/repos/mantidproject/mantid/contents/instrument
GitHubApiHelper-[Debug] Answer from web: 200 OK
GitHubApiHelper-[Debug] GitHub API 4986 of 5000 calls left. Resets at 2018-Dec-17 18:00:06 GMT
DownloadInstrument-[Debug] 0 Files deleted.
DownloadInstrument-[Notice] Downloading 1 file from the instrument repository
DownloadInstrument-[Information] Downloading "/home/pf9/.mantid/instrument/ALF_Definition.xml" from "https://github.com/mantidproject/mantid/blob/master/instrument/ALF_Definition.xml?raw=1"
InternetHelper-[Debug] DownloadFile : https://github.com/mantidproject/mantid/blob/master/instrument/ALF_Definition.xml?raw=1 to file: /home/pf9/.mantid/instrument/ALF_Definition.xml
InternetHelper-[Debug] Sending https GET request to: https://github.com/mantidproject/mantid/blob/master/instrument/ALF_Definition.xml?raw=1
GitHubApiHelper-[Debug] Answer from web: 302 Found
InternetHelper-[Debug] url relocated to https://github.com/mantidproject/mantid/raw/master/instrument/ALF_Definition.xmlSending https GET request to: https://github.com/mantidproject/mantid/raw/master/instrument/ALF_Definition.xml
GitHubApiHelper-[Debug] Answer from web: 302 Found
InternetHelper-[Debug] url relocated to https://raw.githubusercontent.com/mantidproject/mantid/master/instrument/ALF_Definition.xmlSending https GET request to: https://raw.githubusercontent.com/mantidproject/mantid/master/instrument/ALF_Definition.xml
GitHubApiHelper-[Debug] Answer from web: 200 OK
DownloadInstrument-[Debug] Time to validate properties: 0.000004 seconds
```

jfkcooper, 2019-01-04:
It may (or may not) be related, but I believe that the same thing may occur in the windows builds. I have generally found that it occurs when the computer "thinks" it is connected to the internet, i.e. there is a live network connection, but this connection does not allow Mantid to download the instruments. In Windows (without logging) this just appears as a crash on startup, reopening Mantid will then "fix" this,

DanNixon, 2019-01-04:
@marshallmcdonnell Your use case sounds like what I was doing when I found https://github.com/mantidproject/mantid/issues/23517.

If the unit test creates and destroys the framework in less time than the instrument updates take then you always end in a segfault.

FWIW I also cannot reproduce the failure with no/unresponsive internet. I'm also not sure where the lack of internet connection comes into this, if the original problem was found on CI infrastructure then having an active internet connection is a pretty safe bet?

peterfpeterson, 2019-01-04:
The hypothesis of lack of internet connection was based on the assumption that travis would severely limit what internet access was available. In trying to sort it out, I discovered that the logging was lacking (and some other modernization) so I created #24352. It was an attempt to debug things by improving logging to see if it would point out where the issues actually were. 

What @DanNixon describes is a bit more credible and more consistent with (I forget the correct term) an invalidated pointer. I remember @martyngigg fixing one recently in #24229, but I'm not sure where to fix/how to create a test in this particular case.

marshallmcdonnell, 2019-01-04:
So just  a few comments:

1) I first encountered it via using Travis-CI but was reproducing it locally by running the docker images interactively, starting python (both mantidpython and just python with mantid in the PYTHONPATH), and then issuing `import mantid.simpleapi`. So using travis was a "red herring".

2) Trying it now, I can no longer reproduce the error either. Three reasons I can think of this: 1) I did up date my docker version so maybe this was a docker issue and 2) something in Mantid Framework has been merged in that fixed this issue or 3) I had some "special" circumstances that day. All are speculation, though.

Since this is no longer reproducible from the initial report, think it is okay to just close the issue for now @peterfpeterson @DanNixon @jfkcooper ?

closed 2019-01-04 by marshallmcdonnell