mantidproject/mantid #24449: Editor doesn't point at correct line being executed

peterfpeterson, 2019-01-09:
Current mantidplot installs a trace function so the line
https://github.com/mantidproject/mantid/blob/master/MantidPlot/src/PythonScript.cpp#L57
emits the line number and the editor properly points at what is executing. 

In mantidworkbench the equivalent is 
https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/widgets/codeeditor/execution.py#L157
which does not point at the correct line.

martyngigg, 2019-01-09:
I would argue that the workbench code is working as intended, pointing at a block but not a line, but I will accept the tracing function is  more accurate. We should check the performance though before running with it as it involves a callback on every single line of Python code that is executed.

peterfpeterson, 2019-01-09:
There are two parts on this one: it isn't always pointing at the correct block; the code I'm working with has a loop over files/materials and is 55 lines long. Pointing at the `for` isn't precise enough. Once the code has been converted to production, the loop should be much smaller.

PR #27986 (2020-02-21): Replace block-style progress reporter with C-api based one
**Description of work.**

Replaces the script editor execution mechanism with a C++ based one. See commit message for more details. It greatly improves the accuracy of the progress reporter.

**Report to:** James Lord/ISIS

**To test:**

Test out as many scripts as you can. Try aborting, creating syntax errors, runtime errors and try all of the scripts in the issues below.

Fixes #27559
Fixes #27713
Fixes #27245 
Fixes #24449
fixes #27923
---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mant [... cut]
closed 2020-02-21 by NickDraper