mantidproject/mantid #24570: Indirect Elwin - Ability to enter Run numbers like Energy Transfer

robertapplin, 2019-01-23:
**Original reporter:** Spencer

### Expected behavior
In the Elwin interface, it would be useful to be able to enter Run numbers when loading in data. This will make it easier to have multiple input.

### Steps to reproduce the behavior
1.`Interfaces`->`Indirect`->`Data Analysis`->`Elwin` interface

### Platforms affected
All

AnthonyLim23, 2019-08-14:
can you show me this please

robertapplin, 2019-08-15:
@SpencerHowells Would this use the run numbers to calculate EnergyTransfer before running Elwin?

SpencerHowells, 2019-08-15:
@AnthonyLim23 This is an old Issue for the current ElWin which use reduced data files. The idea was to be able to input run-numbers like EnergyTransfer and it would create the reduced file names. In my 'incredible plan' the current Elwin would move to the new Manipulation interface & the Reduction interface would have a version which uses raw data & start off by running EnergyTransfer.

AnthonyLim23, 2021-03-01:
@SpencerHowells can we close this in place of the planned reorganisation?

SpencerHowells, 2021-03-01:
@AnthonyLim23 Yes- will be in reorganisation

closed 2021-09-04 by stale[bot]