mantidproject/mantid #24576: Algorithm to wrap real space pystog functions

peterfpeterson, 2019-01-23:
The algorithm should have an input workspace, output workspace, input workspace function type (`G`, `GK`, `g` but names can be changed as appropriate) and call all the functions [line 210 to 297](https://github.com/marshallmcdonnell/pystog/blob/master/pystog/converter.py#L210). When needed, the scattering information should be taken from the sample object of the workspace.

This requires #24442 to be done first.

marshallmcdonnell, 2019-01-29:
Test data can be found [here](https://raw.githubusercontent.com/marshallmcdonnell/pystog/master/data/test_data/argon.real_space.dat)

And here is an example of:
1. Loading test data, specifically g(r), into a Mantid workspace
2. Unpack workspace
3. Use PyStoG to convert to other real space functions and plot

```python
import wget
import numpy as np
import matplotlib.pyplot as plt
from mantid.simpleapi import CreateWorkspace 
from pystog import Converter

# Grab the reciprocal data for argon
url = "https://raw.githubusercontent.com/marshallmcdonnell/pystog/master/data/test_data/argon.real_space.dat"
filename = wget.download(url)
r, gofr, GofR_, GKofR_ = np.loadtxt(filename, skiprows=2, unpack=True)

# Convert S(Q) to Mantid wksp 
wksp = CreateWorkspace(DataX=r, DataY=gofr,
                       UnitX="Angstrom",
                       Distribution=True)

# Convert reciprocal-space functions
converter = Converter()
r = wksp.readX(0)
g = wksp.readY(0)
G = converter.g_to_G(r, g, **{'rho': 2.0})
GK = converter.g_to_GK(r, g, **{'rho': 2.0, '<b_coh>^2': 2.0})

# Plot results
ax = plt.subplot(111, projection='mantid')
ax.plot(r, g, label='g(r)')
ax.plot(r, G, label='G(r)')
ax.plot(r, GK, label='GK(r)')
ax.legend()
plt.show()
```

marshallmcdonnell, 2019-01-30:
Proposed name of the algorithm `PDConvertRealSpace`

closed 2019-03-03 by AndreiSavici