mantidproject/mantid #24577: Algorithm to wrap reciprocal to real space pystog functions

peterfpeterson, 2019-01-23:
The algorithm should have an input workspace, output workspace, input workspace function type (`S`, `F`, `FK`, `DCS` but names can be changed as appropriate) and call all the functions [line 183 to 403](https://github.com/marshallmcdonnell/pystog/blob/master/pystog/transformer.py#L183). When needed, the scattering information should be taken from the sample object of the workspace.

This should be a replacement for [PDFFourierTransform](http://docs.mantidproject.org/nightly/algorithms/PDFFourierTransform-v1.html)

This requires #24442 to be done first.

marshallmcdonnell, 2019-01-29:
Test data can be found [here](https://raw.githubusercontent.com/marshallmcdonnell/pystog/master/data/test_data/argon.reciprocal_space.dat)

And here is an example of:
1. Loading test data into a Mantid workspaces
2. Unpack workspaces
3. Use PyStoG to fourier transform each reciprocal space function into all available real space functions and plot

```python
import wget
import numpy as np
import matplotlib.pyplot as plt
from mantid.simpleapi import CreateWorkspace 
from pystog import Transformer 

# Grab the reciprocal data for argon
url = "https://raw.githubusercontent.com/marshallmcdonnell/pystog/master/data/test_data/argon.reciprocal_space.dat"
filename = wget.download(url)
q, SofQ, FofQ, FKofQ, DCSofQ = np.loadtxt(filename, skiprows=2, unpack=True)

# Convert S(Q) to Mantid wksp 
sq_wksp = CreateWorkspace(DataX=q, DataY=SofQ,
                       UnitX="MomentumTransfer",
                       Distribution=True)
fq_wksp = CreateWorkspace(DataX=q, DataY=FofQ,
                       UnitX="MomentumTransfer",
                       Distribution=True)
fk_wksp = CreateWorkspace(DataX=q, DataY=FKofQ,
                       UnitX="MomentumTransfer",
                       Distribution=True)
dcs_wksp = CreateWorkspace(DataX=q, DataY=DCSofQ,
                       UnitX="MomentumTransfer",
                       Distribution=True)

# Unpack workpsaces
q = sq_wksp.readX(0)
sq = sq_wksp.readY(0)
fq = fq_wksp.readY(0)
fk = fk_wksp.readY(0)
dcs = dcs_wksp.readY(0)

# Transform from each reciprocal-space functions
transformer = Transformer()
r = np.arange(0.0, 50.0, 0.02)
kwargs = {"rho": 0.02138, "<b_coh>^2": 3.644, "<b_tot^2>": 5.435}

r, g_from_sq = transformer.S_to_g(q, sq, r, **kwargs)
r, G_from_sq = transformer.S_to_G(q, sq, r)
r, GK_from_sq = transformer.S_to_GK(q, sq, r, **kwargs)

r, g_from_fq = transformer.F_to_g(q, fq, r, **kwargs)
r, G_from_fq = transformer.F_to_G(q, fq, r)
r, GK_from_fq = transformer.F_to_GK(q, fq, r, **kwargs)

r, g_from_fk = transformer.FK_to_g(q, fk, r, **kwargs)
r, G_from_fk = transformer.FK_to_G(q, fk, r, **kwargs)
r, GK_from_fk = transformer.FK_to_GK(q, fk, r, **kwargs)

r, g_from_dcs = transformer.DCS_to_g(q, dcs, r, **kwargs)
r, G_from_dcs = transformer.DCS_to_G(q, dcs, r, **kwargs)
r, GK_from_dcs = transformer.DCS_to_GK(q, dcs, r, **kwargs)

# Plot results
ms=2
fig, ax = plt.subplots(3, 1, subplot_kw={'projection':'mantid'})
ax[0].plot(r, g_from_sq,  '-',  markersize=ms, label='g(r) from S(Q)')
ax[0].plot(r, g_from_fq,  '--', markersize=ms, label='g(r) from F(Q)')
ax[0].plot(r, g_from_fk,  'o',  markersize=ms, label='g(r) from FK(Q)')
ax[0].plot(r, g_from_dcs, 'x',  markersize=ms, label='g(r) from DCS(Q)')
ax[0].legend()

ax[1].plot(r, G_from_sq,  '-',  markersize=ms, label='G(r) from S(Q)')
ax[1].plot(r, G_from_fq,  '--', markersize=ms, label='G(r) from F(Q)')
ax[1].plot(r, G_from_fk,  'o',  markersize=ms, label='G(r) from FK(Q)')
ax[1].plot(r, G_from_dcs, 'x',  markersize=ms, label='G(r) from DCS(Q)')
ax[1].legend()

ax[2].plot(r, GK_from_sq,  '-',  markersize=ms, label='GK(r) from S(Q)')
ax[2].plot(r, GK_from_fq,  '--', markersize=ms, label='GK(r) from F(Q)')
ax[2].plot(r, GK_from_fk,  'o',  markersize=ms, label='GK(r) from FK(Q)')
ax[2].plot(r, GK_from_dcs, 'x',  markersize=ms, label='GK(r) from DCS(Q)')
ax[2].legend()
plt.show()
```

marshallmcdonnell, 2019-01-30:
Proposed name of the algorithm `PDTransformReciprocalToReal`

marshallmcdonnell, 2021-02-23:
WIP

peterfpeterson, 2021-08-23:
WIP

closed 2022-02-26 by stale[bot]