mantidproject/mantid #24578: Algorithm to wrap real to reciprocal space pystog functions

peterfpeterson, 2019-01-23:
The algorithm should have an input workspace, output workspace, input workspace function type (`G`, `GK`, `g` but names can be changed as appropriate) and call all the functions [line 407 to 616](https://github.com/marshallmcdonnell/pystog/blob/master/pystog/transformer.py#L407). When needed, the scattering information should be taken from the sample object of the workspace.

marshallmcdonnell, 2019-01-29:
Test data can be found [here](https://raw.githubusercontent.com/marshallmcdonnell/pystog/master/data/test_data/argon.real_space.dat)

And here is an example of:
1. Loading test data into a Mantid workspaces
2. Unpack workspaces
3. Use PyStoG to fourier transform each real space function into all available reciprocal space functions and plot

'''python
import wget
import numpy as np
import matplotlib.pyplot as plt
from mantid.simpleapi import CreateWorkspace 
from pystog import Transformer 

# Grab the real data for argon
url = "https://raw.githubusercontent.com/marshallmcdonnell/pystog/master/data/test_data/argon.real_space.dat"
filename = wget.download(url)
q, gofr, GofR, GKofR = np.loadtxt(filename, skiprows=2, unpack=True)

# Convert real space data to Mantid workspaces
gofr_wksp = CreateWorkspace(DataX=q, DataY=gofr,
                       UnitX="Angstrom",
                       Distribution=True)
GR_wksp = CreateWorkspace(DataX=q, DataY=GofR,
                       UnitX="Angstrom",
                       Distribution=True)
GK_wksp = CreateWorkspace(DataX=q, DataY=GKofR,
                       UnitX="Angstrom",
                       Distribution=True)

# Unpack workpsaces
r = gofr_wksp.readX(0)
g = gofr_wksp.readY(0)
G = GR_wksp.readY(0)
GK = GK_wksp.readY(0)

# Transform from each real-space function to all reciprocal-space functions
transformer = Transformer()
q = np.arange(0.0, 30.0, 0.02)
kwargs = {"rho": 0.02138, "<b_coh>^2": 3.644, "<b_tot^2>": 5.435}

q, sq_from_g = transformer.g_to_S(r, g, q, **kwargs)
q, fq_from_g = transformer.g_to_F(r, g, q, **kwargs)
q, fk_from_g = transformer.g_to_FK(r, g, q, **kwargs)
q, dcs_from_g = transformer.g_to_DCS(r, g, q, **kwargs)

q, sq_from_G = transformer.G_to_S(r, G, q)
q, fq_from_G = transformer.G_to_F(r, G, q)
q, fk_from_G = transformer.G_to_FK(r, G, q, **kwargs)
q, dcs_from_G = transformer.G_to_DCS(r, G, q, **kwargs)

q, sq_from_GK = transformer.GK_to_S(r, GK, q, **kwargs)
q, fq_from_GK = transformer.GK_to_F(r, GK, q, **kwargs)
q, fk_from_GK = transformer.GK_to_FK(r, GK, q, **kwargs)
q, dcs_from_GK = transformer.GK_to_DCS(r, GK, q, **kwargs)

# Plot results
ms=2
fig, ax = plt.subplots(4, 1, subplot_kw={'projection':'mantid'})

ax[0].plot(q, sq_from_g,  '-',  markersize=ms, label='S(Q) from g(r)')
ax[0].plot(q, sq_from_G,  '--', markersize=ms, label='S(Q) from G(r)')
ax[0].plot(q, sq_from_GK,  'o',  markersize=ms, label='S(Q) from GK(r)')
ax[0].legend()

ax[1].plot(q, fq_from_g,  '-',  markersize=ms, label='F(Q) from g(r)')
ax[1].plot(q, fq_from_G,  '--', markersize=ms, label='F(Q) from G(r)')
ax[1].plot(q, fq_from_GK,  'o',  markersize=ms, label='F(Q) from GK(r)')
ax[1].legend()

ax[2].plot(q, fk_from_g,  '-',  markersize=ms, label='FK(Q) from g(r)')
ax[2].plot(q, fk_from_G,  '--', markersize=ms, label='FK(Q) from G(r)')
ax[2].plot(q, fk_from_GK,  'o',  markersize=ms, label='FK(Q) from GK(r)')
ax[2].legend()

ax[3].plot(q, dcs_from_g,  '-',  markersize=ms, label='DCS(Q) from g(r)')
ax[3].plot(q, dcs_from_G,  '--', markersize=ms, label='DCS(Q) from G(r)')
ax[3].plot(q, dcs_from_GK,  'o',  markersize=ms, label='DCS(Q) from GK(r)')
ax[3].legend()
plt.show()
'''

marshallmcdonnell, 2019-01-30:
Proposed name of the algorithm `PDTransformRealToReciprocal`

marshallmcdonnell, 2021-02-23:
WIP

peterfpeterson, 2021-08-23:
WIP

Kvieta1990, 2022-02-21:
fine with me

closed 2023-06-12 by Kvieta1990